swri-robotics/mapviz #644: No transform between /wgs84 and far_field

Mdominik, 2019-08-03:
I'm trying to run mapviz by following danielsnider's tutorial but I keep getting "Status: No transform between /wgs84 and far_field". Whenever I hover over the rectangle where the map is supposed to be, I cannot see any longitude and latitude at the bottom. 
I'm running it on Ubuntu 16.04 64-bit. Any help would be appreciated.

pjreed, 2019-08-06:
Transforming to and from `/wgs84` is handled specially; in order for it to work, something needs to be publishing a topic at `/local_xy_origin` that contains a `geometry/_msgs/PoseStamped` message that indicates the WGS84 coordinate that corresponds to (0,0,0) in your discontinuous world-fixed frame (see https://www.ros.org/reps/rep-0105.html); the ROS standard name for this is `/map`, but older versions of mapviz and some of our packages used `/far_field`, so you might still see that in documentation (let me know if you do and I'll make a note to clean it up).

Typically, this is handled by this node in the `mapviz.launch` file:
```xml
<node pkg="swri_transform_util" type="initialize_origin.py" name="initialize_origin">
    <param name="local_xy_frame" value="/map"/>
    <param name="local_xy_origin" value="static_origin"/>
    <rosparam param="local_xy_origins">
        [{ name: static_origin,
         latitude: 37.9879772, <!-- change this -->
         longitude: 23.9078602, <!-- change this -->
         altitude: 129.69,
         heading: 0.0}]
    </rosparam>
</node>
```

The above block will run a node named `initialize_origin` that will publish `/local_xy_origin` with the indicated latitude/longitude.  Alternately, if the `value` for the `local_xy_origin` param is set to `auto`, the node will subscribe to the `gps` and `fix` topics and use the first valid `gps_common/GPSFix` or `sensor_msgs/NavSatFix` message it receives as the origin.

So, when you see an error similar to the one you're seeing, some useful things to check are:

- Is `/local_xy_origin` being published?
- Are the coordinate and frame in it the values you expect?
- Does the Fixed Frame you've selected in mapviz match the local XY frame in that message?
- If you've set `use_sim_time` parameter to true, are you actually playing a bag file?  (If nothing is actively playing, mapviz won't be able get the `/local_xy_origin` message)

pepas, 2019-09-09:
I have the same problem. I checked all the things but it still says: "No transform between /wgs84 and world." I have the same tf tree like this guy: https://answers.ros.org/question/261580/how-do-i-to-make-wgs84-frame-available/ and I just want to see GPS position of my robot on the tile map. When I run mapviz alone, tile map shows. Any help would be appreciated.

pjreed, 2019-09-09:
What do you see when you run `rostopic echo /local_xy_origin`?  Is there a difference between when you're running it on your robot and when you're running mapviz alone?

pepas, 2019-09-11:
There wasn't difference (in both cases valid GPS coordinates) but I solved it yesterday. I use virtualbox and there was problem in network settings.

daniel-dsouza, 2019-09-16:
Marking this issue as resolved.

closed 2019-09-16 by daniel-dsouza