swri-robotics/mapviz #647: ROS Melodic gps_common/GPSFix.h No such file or directory

sorga, 2019-09-15:
During compilation by catkin error:
```
[  2%] Built target swri_system_util
/home/sorga/catkin_ws/src/gps_umd/gpsd_client/src/client.cpp:2:10: fatal error: gps_common/GPSFix.h: No such file or directory
    2 | #include <gps_common/GPSFix.h>
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [gps_umd/gpsd_client/CMakeFiles/gpsd_client.dir/build.make:63: gps_umd/gpsd_client/CMakeFiles/gpsd_client.dir/src/client.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:8653: gps_umd/gpsd_client/CMakeFiles/gpsd_client.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Scanning dependencies of target _marti_nav_msgs_generate_messages_check_deps_SetRoute

```
Inside of gps_common folder only conversion.h file. I've try from differnet git repo clone gps_umd, problem always the same: GPSFix.h: No such file or directory
Is it problem because of apsense of GPS reciever ? Any idea ? Need help.

daniel-dsouza, 2019-09-15:
hello @sorga 

Could you please try installing the `gps_common` package, available from packages.ros.org?

`sudo apt install ros-melodic-gps-common`

sorga, 2019-09-16:
> hello @sorga
> 
> Could you please try installing the `gps_common` package, available from packages.ros.org?
> 
> `sudo apt install ros-melodic-gps-common`

I'm using Archlinux, apt install not work. But inside of ubuntu package ros-melodic-gps-common I found missing files: GPSFix.h and GPSStatus.h. Substituing them to include/gps_common/ seems to solve problem.

daniel-dsouza, 2019-09-16:
Hmm. I would think that adding the [gps_umd repo](https://github.com/swri-robotics/gps_umd) to your catkin workspace would have also worked, and that mapviz would build `swri_transform_util` and its dependency `gps_common` before attempting to build mapviz, as per the package.xml. but I'm glad you found a solution. 

Closing the issue.

closed 2019-09-16 by daniel-dsouza