swri-robotics/mapviz #653: Map does not follow target frame

ShawnSchaerer, 2019-10-18:
I am trying to get the map to follow my target frame but it does not.
Not sure if we are doing something wrong but our TF tree looks correct.

I am getting two errors
 Error: No transform between /wgs84 and map
 Error: No transform between  and map

Any ideas?  Its not really clear on how to make this work and the other issues don;t seem to have a good solution.  
Last point, in RVIZ I can get the map to follow the target frame so it seems its a mapviz config problem.

Thanks

pjreed, 2019-10-18:
So, in this first case:
>  Error: No transform between /wgs84 and map

`/wgs84` is not a standard ROS frame and won't (and shouldn't) show up in your TF tree.  Here's another issue with some info about how to set that up and how to diagnose it if it's not working right: https://github.com/swri-robotics/mapviz/issues/644

Note that if you're not trying to display map tiles or other data with WGS84 coordinates, you don't need that at all.  In a typical use case, you want to set your Fixed Frame to `map` and set your Target Frame to something relative to your vehicle, such as `base_link`.

As for this:
>  Error: No transform between and map

Make sure you don't have a space (` `) in your target frame.  If you completely clear out the contents of the target frame, it should display `<none>`.

ShawnSchaerer, 2019-10-18:
for the /wgs84 error.  I can see the map that is not a problem.
so my fixed frame is called "map" and my target frame is called "gps"
I don't see any spaces in "map" so I am not sure.

ShawnSchaerer, 2019-10-18:
the second error is coming from the robot image plugin, but I can see the robot image and the robot image follows my TF of GPS.
I can live with the errors, but I really want the map to follow the target frame

pjreed, 2019-10-18:
Did you set a Frame in your robot_image plugin?  That error should go away if mapviz can transform between your Target Frame and the robot_image frame.

If you run `rosrun rqt_tf_tree rqt_tf_tree`, does it look like there's a transformation between your `map` and `gps` frames?

ShawnSchaerer, 2019-10-18:
I should have clarified that I only get the messages once on startup.  Probably an initialization issue.

The map following  the target frame is my real concern.  Any ideas?

malban, 2019-10-18:
What is your fixed frame?
What is your target frame?
When you hover your mouse over the map, do you see the lat/lon coordinates in the lower right corner like this?

[image]

Can you post the output pdf of `rosrun tf view_frames` ?
Can you post your mapviz config file?

ShawnSchaerer, 2019-10-19:
My fixed frame is map and my target frame is gps and I have a transform between the two.
yes when I hover over the map with the mouse I get lat/lon coordinates and they update when I move the mouse.
I have attached my mapviz config and the tf-tree
[image]

closed 2019-10-19 by ShawnSchaerer