swri-robotics/mapviz #654: ROS2 Support

ShawnSchaerer, 2019-10-23:
Are there plans to support ROS2 in Mapviz?

matt-attack, 2019-10-24:
Sadly yes.

pjreed, 2019-10-24:
We've got some vague plans, but it is probably still a ways off in the future at this point.  Aside from being a significant undertaking in itself, we're still working our way through all of its dependencies; I'm actually working on porting [marti_common](https://github.com/swri-robotics/marti_common/pull/549) to ROS2 at the moment.

ShawnSchaerer, 2019-10-25:
Thanks for the update.  We might be able to help with the port.

pjreed, 2020-03-05:
We're now working on this in #672 .  Still got a ways to go, but the base mapviz package and several plugins have been ported.

pjreed, 2020-10-22:
I forgot to close this issue out, but Mapviz has been released into ROS Dashing and Eloquent for a while now, and I just released it into Foxy.

closed 2020-10-22 by pjreed