swri-robotics/mapviz #655: robot_image and gps messages

Maintown, 2019-10-28:
Hello everyone,
i am quiet new to ROS and its features and currently I am working on using the robot_image plugin for mapviz. I know that robot_image plugin takes a tf_frame to be fixed to but i read in other threads that it is possible to fix it to a GPS position which is published on a topic. Currently I just dont really get how to configure my launch file to get a frame from the gps message and fix the robot_image to. My tile map is displayed well, but i dont get how i should run the gps_transform_publisher and what parameters i have to set to fix my robot_image to my gps position. My current launch file looks like this:

```
<launch>
    <arg name="node_start_delay" default="2.0"/>
    <node pkg="mapviz" type="mapviz" name="mapviz"/>

    <node pkg="tf" type="static_transform_publisher" name="swri_transform" args="0 0 0 0 0 0 /map /origin 100" launch-prefix="bash -c 'sleep $(arg node_start_delay); $0 $@'"/>

    <node pkg="swri_transform_util" type="initialize_origin.py" name="initialize_origin" launch-prefix="bash -c 'sleep $(arg node_start_delay); $0 $@'">
        <param name="local_xy_frame" value="/map"/>
        <param name="local_xy_origin" value="flprefered"/>
        <rosparam param="local_xy_origins">
          [{ name: flprefered,
            latitude: 52.319167,
            longitude: 10.558889,
            altitude: 84,
            heading: 0.0},
     
           { name: flother,
            latitude: 52.368750,
            longitude: 10.351472,
            altitude: 0.0,
            heading: 0.0}]
        </rosparam>
    </node>

    <node pkg="swri_transform_util" type="initialize_origin.py" name="gps_transform_publisher">
        <param name="local_xy_frame" value="/map"/>
        <param name="local_xy_origin" value="auto"/>
        <rosparam param="local_xy_origins">
        </rosparam>
    </node>
</launch>
```

With my uav I am currently not using tf due to the project I am participating in.

I hope anyone can help me.
Thank you and best regards!

malban, 2019-10-28:
This part of your launch file is wrong:
```
    <node pkg="swri_transform_util" type="initialize_origin.py" name="gps_transform_publisher">
        <param name="local_xy_frame" value="/map"/>
        <param name="local_xy_origin" value="auto"/>
        <rosparam param="local_xy_origins">
        </rosparam>
    </node>
```

`type="initialize_origin.py"` should be `type="gps_transform_publisher"`

You'll need to look at the code for gps_transform_publisher to see what parameters you need.  Here is the relevant section: https://github.com/swri-robotics/marti_common/blob/master/swri_transform_util/src/nodelets/gps_transform_publisher.cpp#L64
```
void GpsTransformPublisher::onInit()
{
  ros::NodeHandle priv = getPrivateNodeHandle();
  swri::param(priv,"child_frame_id", veh_frame_id_, std::string("base_link"));
  swri::param(priv,"parent_frame_id", global_frame_id_, std::string("map"));

  gps_sub_ = getNodeHandle().subscribe("gps", 100, &GpsTransformPublisher::HandleGps, this);

  tf_manager_.Initialize();
}
```
I think this should work with the default parameters given the rest of your launch file.  It will publish a TF transform between `map` and  `base_link`, where `map` is the global frame and `base_link` is following the GPS position.

You will need to remap the `gps` topic if your GPS topic is not the same:
```
    <node pkg="swri_transform_util" type="gps_transform_publisher" name="gps_transform_publisher">
        <remap from="gps" to="your_gps_topic"/>
    </node>
```

Maintown, 2019-10-28:
I guess this would work, but somehow my system does not seem to have a gps_transform_publisher executable and I just updated the swri_transform_util package for kinetic. I just searched for it and its not there.

malban, 2019-10-28:
Looks like `gps_transform_publisher` is not installed in the `CMakeLists.txt`, so it is missing from the debian package.

A workaround would be to add `swri_transform_util` to your workspace as a source package.

malban, 2019-10-28:
https://github.com/swri-robotics/marti_common/issues/550

Maintown, 2019-10-29:
It worked perfecly! Thank you very much!
This issue can be closed then.
In short terms for everyone having trouble with this:
I cloned the repo for swri_transform_util and put it in my src folder of my ws, built the whole thing and in my launch file i set the parameters for child and parent frame ids. To make this happen I used these launchfile settings:

```
<launch>
    <node pkg="swri_transform_util" type="gps_transform_publisher" name="gps_transform_publisher" launch-prefix="bash -c 'sleep $(arg node_start_delay); $0 $@'">
        <param name="parent_frame_id" value="map"/>
        <param name="child_frame_id" value="uav"/>
    </node>

    <node pkg="sensor_conversion" type="sensor_conversion_node" name="sensor_conversion_node"/>
</launch>
```
I set the frame for the robot image to uav and it followed the positon perfectly.
Thanks again and have a good day!

pjreed, 2019-10-31:
Thanks, glad to see this is all resolved!  I merged in a PR to install `gps_transform_publisher` -- although note that it's also a nodelet, and the nodelet library is installed, so you should be able to run it as a nodelet from the existing debian package.