swri-robotics/mapviz #660: Glew error __glewVertexBindingDivisor

waddellt15, 2019-11-26:
I get this error when trying catkin_make, not sure what the root of it is.

In file included from /home/badgerboat/catkin_ws/src/mapviz/mapviz/src/map_canvas.cpp:31:0:
/usr/include/GL/glew.h:17830:46: note: previous declaration ‘void (* __glewVertexAttribBinding)(GLuint, GLuint)’
 GLEW_FUN_EXPORT PFNGLVERTEXATTRIBBINDINGPROC __glewVertexAttribBinding;
                                              ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/aarch64-linux-gnu/qt5/QtGui/qopengl.h:105:0,
                 from /usr/include/aarch64-linux-gnu/qt5/QtOpenGL/qgl.h:45,
                 from /usr/include/aarch64-linux-gnu/qt5/QtOpenGL/QGLWidget:1,
                 from /home/badgerboat/catkin_ws/src/mapviz/mapviz/include/mapviz/map_canvas.h:42,
                 from /home/badgerboat/catkin_ws/src/mapviz/mapviz/src/map_canvas.cpp:35:
/usr/include/GLES3/gl31.h:1520:88: error: ‘void __glewVertexBindingDivisor(GLuint, GLuint)’ redeclared as different kind of symbol
  void GL_APIENTRY glVertexBindingDivisor (GLuint bindingindex, GLuint divisor);

pjreed, 2019-11-26:
From your include paths, I'm guessing you're building on an ARM64 architecture?  Unfortunately, mapviz doesn't currently build on ARM64; also see the build job on the official build farm: http://build.ros.org/job/Mbin_ubv8_uBv8__mapviz__ubuntu_bionic_arm64__binary/

I believe it's due to QtOpenGL using OpenGL APIs that aren't supported on the platform, but I don't have access to a platform I can test it on.  If there is anybody out there who could get it working, it would be greatly appreciated.

danthony06, 2019-11-26:
I've got an Odroid and RPi at home that I can try it out on.

waddellt15, 2019-11-26:
I am using a Jetson Nano with their unique ubuntu 16.0.4 kernel if that helps? Let me know if there is anything I can to do help

danthony06, 2019-11-26:
Okay, we've got one of those. I'll try to find some time to work on it.

danthony06, 2019-12-09:
Sorry for the delayed response, I tried getting ROS going on my Pi this
weekend and had some issues that I'm working through. I haven't forgotten
about this problem.

waddellt15, 2019-12-16:
Awesome thank you!

danthony06, 2020-01-01:
[build_log.txt](https://github.com/swri-robotics/mapviz/files/4014476/build_log.txt)

For reference, here is the compete output when the build fails on my Odroid running 18.04 and ROS Melodic.

danthony06, 2020-01-19:
@pjreed I think the only reasonable solution to this is to remove GLEW and use Qt's OpenGL support instead. It's likely to be a substantial change, and maybe we want to open up an issue specific to that instead.

matt-attack, 2020-01-20:
Theoretically it should just be changing some include file names. It seems that the Qt one is already being included somehow, so it could just be deleting some.

matt-attack, 2020-01-20:
So, deleting the includes mostly works in that spot. Though, we are apparently using an OpenGL extension so we might have to manually import those functions or find them in the mess that is Qt's approach of importing OpenGL https://www.kdab.com/opengl-in-qt-5-1-part-1/

matt-attack, 2020-01-20:
After poking around, it seems this extension isn't needed on OpenGL >= 2.1. So we could just do a version check and use the normal functions.

matt-attack, 2020-01-20:
@danthony06 you could try this branch https://github.com/matt-attack/mapviz/tree/remove-extension-test

It needs cleaning up, but doesn't seem to break anything (other than the extensions which are just disabled atm) and avoids the double declaration issue. Was only able to try it on WSL Ubuntu on my laptop so someone would need to try it on a Pi

danthony06, 2020-01-20:
Trying the build now. I'm not sure what the best way to test everything is,
other than to see if it runs. I don't have many bag files at home that
would be useful for testing this on, but I can bring in my Odroid tomorrow.

danthony06, 2020-01-20:
Never mind, it's failing to build. Attaching the build output.
[build_fail.txt](https://github.com/swri-robotics/mapviz/files/4084379/build_fail.txt)

danthony06, 2020-01-20:
I started including GL/gl.h, and it's fixing some build errors, but it looks like all of the plugins will need a corresponding change to at least build.

danthony06, 2020-01-20:
@matt-attack I forked your repo into my account and added various includes. I'm still getting the following build errors, but I'll try to look at it again later. It looks like we're not linking against the correct OpenGL libs now.

```
/home/danthony/src/mapviz/devel/.private/multires_image/lib/libmultires_widget.so: undefined reference to `glLoadIdentity'
/home/danthony/src/mapviz/devel/.private/multires_image/lib/libmultires_image.so: undefined reference to `glVertex2f'
/home/danthony/src/mapviz/devel/.private/multires_image/lib/libmultires_image.so: undefined reference to `glEnd'
/home/danthony/src/mapviz/devel/.private/multires_image/lib/libmultires_image.so: undefined reference to `glBegin'
/home/danthony/src/mapviz/devel/.private/multires_image/lib/libmultires_widget.so: undefined reference to `glMatrixMode'
/home/danthony/src/mapviz/devel/.private/multires_image/lib/libmultires_widget.so: undefined reference to `glOrtho'
/home/danthony/src/mapviz/devel/.private/multires_image/lib/libmultires_image.so: undefined reference to `glTexCoord2f'
/home/danthony/src/mapviz/devel/.private/multires_image/lib/libmultires_widget.so: undefined reference to `glColor4f'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/danthony/src/mapviz/devel/.private/multires_image/lib/multires_image/multires_view_node] Error 1
make[1]: *** [CMakeFiles/multires_view_node.dir/all] Error 2
make: *** [all] Error 2
```

danthony06, 2020-01-20:
Okay, it's building now.

JeremieBourque1, 2020-01-23:
@danthony06 Is there any way this can work with melodic as well?

danthony06, 2020-01-23:
We're working on getting in running on both Kinetic and Melodic at the
present. I don't want to promise anything, but I think we'll get it fixed.