swri-robotics/mapviz #683: Plot multi paths in same component

TianqiLi7398, 2020-05-17:
I'm using mapviz to display the trajectories I detected and tracked. Due to the number of tracks is not fixed, can I put a list a nav_msg/Path into same ROS topic for mapviz to draw? I feel we cannot do that right now.

pjreed, 2020-05-18:
The path plugin is designed similar to rviz's; it expects that every time you publish a new path, it is replacing the old one.

You could so something similar to the suggested answers for rviz; I think it would be very easy to make your node publish an array as `nav_msg/Path[]`, so you publish all of your paths every time, and then modify mapviz's path plugin to also handle arrays.

Another alternative would be to add a history buffer to the Path plugin so that it could store and draw the last N paths it has received, but I think you would want to set a configurable maximum buffer size, since some systems may publish new path messages very frequently with the intent of replacing the previous one.

TianqiLi7398, 2020-05-21:
Thanks for your advice! The following question is about how to change the code for plotting multiple  paths.

Now I make a customized message called `PathArray` which is the array version of nav_msg/Path, and I changed the function in mapviz_plugin/src/path_plugin.cpp ->` PathPlugin::pathCallback`, I put my customized message inside this function. What I actually changed is the following

```
void PathPlugin::pathCallback(const sa_multi_target_tracking::PathArray::ConstPtr& patharray)
{
    if (!has_message_)
    {
      initialized_ = true;
      has_message_ = true;
    }

    ClearPoints();
  for (unsigned int j = 0; j < patharray->paths.size(); j++)
  {
    for (unsigned int i = 0; i < patharray->paths[j].poses.size(); i++)
    {
      StampedPoint stamped_point;
      stamped_point.stamp = patharray->paths[j].header.stamp;
      stamped_point.source_frame = patharray->paths[j].header.frame_id;
      stamped_point.point = tf::Point(patharray->paths[j].poses[i].pose.position.x,
                                      patharray->paths[j].poses[i].pose.position.y, 0);

      pushPoint( std::move(stamped_point) );
    }
  }
  }
```
And my result is the screenshot below, based on my observation it just alines all points in all paths together as one path. So can you point out how to solve this problem. Thanks a lot!
[image]

TianqiLi7398, 2020-05-22:
A follow up of my question. It seems that I solved this problem by changing function `PointDrawingPlugin::DrawLines()` in mapviz_plugin/plot_drawing_plugin.cpp by interrupting `glBegin(GL_LINE_STRIP);` at the end point of each path.

closed 2020-06-16 by TianqiLi7398