swri-robotics/mapviz #685: The latitude and longitude never change in Mapiz

LXlx916, 2020-05-18:
Hi,  I tried to use Mapviz to plot the sensor_msgs/NavSatFix. 
My system is Ubuntu 18.04 with ROS melodic. 

First, I use roslaunch mapviz mapviz.launch to run mapviz. 
I didn't change mapviz.launch so it kept default.

`<launch>

  <node pkg="mapviz" type="mapviz" name="mapviz"></node>

  <node pkg="swri_transform_util" type="initialize_origin.py" name="initialize_origin" >
    <param name="local_xy_frame" value="/map"/>
    <param name="local_xy_origin" value="swri"/>
    <rosparam param="local_xy_origins">
      [{ name: swri,
         latitude: 29.45196669,
         longitude: -98.61370577,
         altitude: 233.719,
         heading: 0.0},
         
       { name: back_40,
         latitude: 29.447507,
         longitude: -98.629367,
         altitude: 200.0,
         heading: 0.0}]
    </rosparam>
  </node>

  <node pkg="tf" type="static_transform_publisher" name="swri_transform" args="0 0 0 0 0 0 /map /origin 100"  />

</launch>`

Then, I add navsat and tile_map in Mapviz like this.

[image]

When I played the bagfile. The latitude and longitude kept the same in Mapviz. I have read some relative Issues before. But I can't understand the reason. Maybe I missed some key steps?

This is my config.
`capture_directory: ~
fixed_frame: map
target_frame: <none>
fix_orientation: false
rotate_90: false
enable_antialiasing: true
show_displays: true
show_status_bar: true
show_capture_tools: true
window_width: 1389
window_height: 759
view_scale: 1
offset_x: -48
offset_y: -10
use_latest_transforms: true
background: "#a0a0a4"
displays:
  - type: mapviz_plugins/navsat
    name: new display
    config:
      visible: true
      collapsed: false
      topic: /navsat/fix
      color: "#00ff00"
      draw_style: lines
      position_tolerance: 20
      buffer_size: 0
  - type: mapviz_plugins/tile_map
    name: new display
    config:
      visible: true
      collapsed: false
      custom_sources:
        []
      bing_api_key: ""
      source: Stamen (terrain)`

Here is my bagfile.
[image]

[GPS_2020-05-13-14-27-39.zip](https://github.com/swri-robotics/mapviz/files/4644825/GPS_2020-05-13-14-27-39.zip)

pjreed, 2020-05-18:
I see a few issues here.  First, try using this for your launch file:
```yaml
<launch>
  <node pkg="mapviz" type="mapviz" name="mapviz" required="true"/>

  <node pkg="swri_transform_util" type="initialize_origin.py" name="initialize_origin" >
    <param name="local_xy_frame" value="/map"/>
    <param name="local_xy_origin" value="auto"/>
    <rosparam param="local_xy_origins">
      [{ name: swri,
         latitude: 29.45196669,
         longitude: -98.61370577,
         altitude: 233.719,
         heading: 0.0},

       { name: back_40,
         latitude: 29.447507,
         longitude: -98.629367,
         altitude: 200.0,
         heading: 0.0}]
    </rosparam>
    <remap from="fix" to="/navsat/fix"/>
  </node>

  <node pkg="tf" type="static_transform_publisher" name="swri_transform" args="0 0 0 0 0 0 /map /origin 100"  />
</launch>
```

There are two important changes here.  The `initialize_origin.py` node takes in a GPS coordinate and uses that to connect the WGS84 coordinate system to your ROS TF tree; setting `<param name="local_xy_origin" value="auto"/>` will make it use the first GPS message it gets instead of using one of the pre-defined origins.  Next, adding `<remap from="fix" to="/navsat/fix"/>` will make it listen to the topic from your bag file.

Also, mapviz draws its plugins in the order that they are listed in the plugin panel.  In your screenshot, since `navsat` is listed first, it will draw that first, and then it will draw the `tile_map` over that, so you would not be able to see any fixes.

Here's a screenshot of what my mapviz looks like with your bag file but with my updated launch file:

[image]

LXlx916, 2020-05-18:
That's cool. Thank you for your prompt reply. 
I have one last question. If I want to keep the new point in the center of the window, I should set the Target Frame?

pjreed, 2020-05-18:
That's right.  Your ROS system should publish a tf frame that is fixed to your vehicle; this is usually called `base_link`.  If you set your Target Frame to that, the window will stay centered on it.

LXlx916, 2020-05-18:
Thank you very much！

closed 2020-05-18 by LXlx916