swri-robotics/mapviz #690: ROS Noetic version

pxlong, 2020-06-01:
Hi, does this package work with ROS Noetic version? If not, do you have a plan to support it?

pjreed, 2020-06-01:
I don't think it will work quite yet, but we do plan to support it.  Right now one of its dependencies is being held up due to needing to migrate to OpenCV 4 (see https://github.com/swri-robotics/marti_common/pull/581), but after that is done, I suspect mapviz should be easy to get working in Noetic.

kdr0588, 2020-07-20:
I'm using Noetic too, but have trouble with the Qt version. I get an error, "_Found unsuitable Qt version "5.12.8" from /usr/bin/qmake, this code requires Qt 4.x_"

pjreed, 2020-10-22:
I actually forgot to close out this issue, but Mapviz has been released in ROS Noetic for a while now.  (I also just released it into ROS Foxy)

closed 2020-10-22 by pjreed

zhangxianzhang, 2021-11-21:
> I'm using Noetic too, but have trouble with the Qt version. When I do catkin_make, I get an error, "_Found unsuitable Qt version "5.12.8" from /usr/bin/qmake, this code requires Qt 4.x_". What would you suggest?

zhangxianzhang, 2021-11-21:
> I'm using Noetic too, but have trouble with the Qt version. When I do catkin_make, I get an error, "_Found unsuitable Qt version "5.12.8" from /usr/bin/qmake, this code requires Qt 4.x_". What would you suggest?

Friends, is there a solution now?

zhangxianzhang, 2021-11-21:
> I'm using Noetic too, but have trouble with the Qt version. When I do catkin_make, I get an error, "_Found unsuitable Qt version "5.12.8" from /usr/bin/qmake, this code requires Qt 4.x_". What would you suggest?

@kdr0588 Friends, is there a solution now?