swri-robotics/mapviz #707: remaping topic

geoeo, 2020-10-12:
Hello,

Is there a list of topic names that I can use to remap from.

According to this tutorial https://medium.com/@hitlx916/visualize-the-gnss-messages-in-mapviz-ros-4ae7eec19936 for NavSatFix it is "fix", which works, but I could not find it in the source code. What are the default topic names for other plugins e.g. image plugin?

Best,
Marc

geoeo, 2020-10-12:
I found https://github.com/swri-robotics/marti_common/blob/master/swri_transform_util/nodes/initialize_origin.py It seems to be defined there. I see there is no default image topic published. How would I go around in doing that? Essentially I want to add an image plugin from a launch file and remap its topic

pjreed, 2020-10-12:
Most of mapviz's plugins don't have default topics; they are blank when added and expect a user to set them through the UI.

If you have a specific set of plugins you always want to use, the best way to do that is to run mapviz normally, configure all of the plugins in the way you want them, and then save that as a configuration file.  You can then pass the path to that config file as an argument to mapviz's `config` ROS param, and it will automatically load that config when it starts.

If you need to be able to configure it in a more dynamic manner, that's a little trickier.  mapviz exposes a ROS service at `add_mapviz_display` that you can call to make it load a plugin, and the `properties` value in the service request can contain a set of key:value pairs that are just like the a plugin's `config` block out of a mapviz config file; so, for example, you could call that service with a `type` of `mapviz_plugins/image` and add a property with the key of `topic` and value of `/camera/image` to make it subscribe to that topic.  I haven't actually tried doing this in a launch file before; I think it'd work, although if you're creating multiple plugins, they might not always get initialized in the right order.

Does either of those methods work for you?

geoeo, 2020-10-13:
I understand. I dont want to spend too much time on this. So a config will do fine. 
Thank you for the quick reply

closed 2020-10-13 by geoeo