swri-robotics/mapviz #712: ROS2 Foxy can't use launch file.

nicolasaw, 2020-12-01:
Hi, I am really new in this Mapviz thing, and I am currently working on ROS Foxy. I figure that Mapviz is now workable on Foxy but somehow when I run the basic launch file, ros2 launch mapviz.launch.py, it does not work and show this output. It seems the parameter setting for the initialization_origin, specifically the local_xy_origins is faulty. Here attached is the output, and I would love to hear some suggestions. THANKS!!
[image]
[image]

pjreed, 2020-12-04:
I hadn't tested this launch file before in Foxy, but I can confirm that it's not working there.  I'll see what needs to be done to fix it.

pjreed, 2020-12-04:
This should be fixed in the `dashing-devel` branch now; let me know if you have any more problems with them.

nicolasaw, 2020-12-05:
Okay thanks, I will try it out. Any idea what causes the problem?

nicolasaw, 2020-12-05:
Hi, so I succeeded in using the launch file. Unfortunately, it states that WGS84Transformer is not initialized. 

[image]

Any idea on this one?

nicolasaw, 2020-12-18:
Hi, sorry its been a while I haven't tried figuring out my problems in Mapviz. I think there is still something wrong with the launch file. As the /local_xy_origin is still not publishing when i run ros2 topic echo. When I checked whether the parameter has been set or not, it shows these 
[image]

When I checked the launch file, it was suppossed to be SWRI and the having set the latitude and longitude. Might this be the cause of what I have right now? 

I also simply can't to set the fixed frame and target frame to map -> origin. This is what my mapviz looks like right now with its config 
[image]

I read this https://github.com/swri-robotics/mapviz/issues/618, but it does not work for me. Any help on this?

nicolasaw, 2020-12-18:
Sorry, it is me again. I somehow found out that my Stamen (terrain) shows something, but it is not complete. It is like this 
[image]

What is wrong with this? Is it my internet connection? I am connected to a 80 Mbps Wifi through LAN

pjreed, 2020-12-18:
First, to explain how this is supposed to work...

In order to draw objects in ROS frames on map tiles, mapviz needs to be able to transform between ROS frames and WGS84 coordinates, but the ROS tf2 system does not natively support doing any types of transformations other than translation and rotation.  To do this, mapviz uses a class called WGS84Transformer.

For WGS84Transformer to work, it needs to know the WGS84 coordinate that corresponds to (0, 0) in ROS's map frame, and we use the `swri_transform_util/initialize_origin.py` node to provide that.  Mapviz's default launch file provides a hard-coded location named `swri`, although on a real robot, you would want to send the output from your GPS to the `initialize_origin.py` node so that it can determine your true origin on startup.

Here's an example of me launching mapviz, and what it looks like immediately after startup:
```
$ ros2 launch mapviz mapviz.launch.xml 
[INFO] [launch]: All log files can be found below /home/preed/.ros/log/2020-12-18-10-08-16-033519-wyvern-51258
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [mapviz-1]: process started with pid [51260]
[INFO] [initialize_origin.py-2]: process started with pid [51262]
[INFO] [static_transform_publisher-3]: process started with pid [51264]
[static_transform_publisher-3] 1608307698.061562 [230] static_tra: using network interface wlp82s0 (udp/192.168.2.109) selected arbitrarily from: wlp82s0, br-1a95fe46b61d, br-63ba9ba7b1aa, docker0, br-ecc4325848b8, tun0
[static_transform_publisher-3] [INFO] [1608307698.170442161] [swri_transform]: Spinning until killed publishing transform from 'map' to 'origin'
[initialize_origin.py-2] 1608307699.604647 [230]    python3: using network interface wlp82s0 (udp/192.168.2.109) selected arbitrarily from: wlp82s0, br-1a95fe46b61d, br-63ba9ba7b1aa, docker0, br-ecc4325848b8, tun0
[initialize_origin.py-2] [INFO] [1608307699.730835394] [initialize_origin]: Origin: swri
[mapviz-1] qt5ct: using qt5ct plugin
[mapviz-1] 1608307699.894098 [230]     mapviz: using network interface wlp82s0 (udp/192.168.2.109) selected arbitrarily from: wlp82s0, br-1a95fe46b61d, br-63ba9ba7b1aa, docker0, br-ecc4325848b8, tun0
[mapviz-1] qt5ct: D-Bus global menu: no
[initialize_origin.py-2] [INFO] [1608307699.965920982] [initialize_origin]: Origin from 'manual' source set to 29.45196669, -98.61370577, 233.719
[mapviz-1] [INFO] [1608307699.984370149] [mapviz]: Target frame selected: <none>
[mapviz-1] [INFO] [1608307700.165779647] [mapviz]: View scale: 1.000000 meters/pixel
[mapviz-1] [INFO] [1608307700.290989578] [mapviz]: Subscribing to /local_xy_origin
[mapviz-1] [ERROR] [1608307700.291262486] [swri_transform_util::Transformer]: UtmTransformer::Initialize: No LocalXyWgs84Util has been set!
[mapviz-1] [ERROR] [1608307700.291281610] [swri_transform_util::Transformer]: Wgs84Transformer::Initialize: local_yx_util was unset!
[mapviz-1] [INFO] [1608307700.339553094] [mapviz]: Found mapviz plugin: mapviz_plugins/attitude_indicator
[mapviz-1] [INFO] [1608307700.339592686] [mapviz]: Found mapviz plugin: mapviz_plugins/coordinate_picker
[mapviz-1] [INFO] [1608307700.339598951] [mapviz]: Found mapviz plugin: mapviz_plugins/disparity
[mapviz-1] [INFO] [1608307700.339604082] [mapviz]: Found mapviz plugin: mapviz_plugins/draw_polygon
[mapviz-1] [INFO] [1608307700.339608716] [mapviz]: Found mapviz plugin: mapviz_plugins/float
[mapviz-1] [INFO] [1608307700.339613504] [mapviz]: Found mapviz plugin: mapviz_plugins/gps
[mapviz-1] [INFO] [1608307700.339618367] [mapviz]: Found mapviz plugin: mapviz_plugins/grid
[mapviz-1] [INFO] [1608307700.339623367] [mapviz]: Found mapviz plugin: mapviz_plugins/image
[mapviz-1] [INFO] [1608307700.339628165] [mapviz]: Found mapviz plugin: mapviz_plugins/laserscan
[mapviz-1] [INFO] [1608307700.339632859] [mapviz]: Found mapviz plugin: mapviz_plugins/marker
[mapviz-1] [INFO] [1608307700.339637647] [mapviz]: Found mapviz plugin: mapviz_plugins/measuring
[mapviz-1] [INFO] [1608307700.339642315] [mapviz]: Found mapviz plugin: mapviz_plugins/multires_image
[mapviz-1] [INFO] [1608307700.339646977] [mapviz]: Found mapviz plugin: mapviz_plugins/navsat
[mapviz-1] [INFO] [1608307700.339651593] [mapviz]: Found mapviz plugin: mapviz_plugins/occupancy_grid
[mapviz-1] [INFO] [1608307700.339656296] [mapviz]: Found mapviz plugin: mapviz_plugins/odometry
[mapviz-1] [INFO] [1608307700.339661029] [mapviz]: Found mapviz plugin: mapviz_plugins/path
[mapviz-1] [INFO] [1608307700.339665741] [mapviz]: Found mapviz plugin: mapviz_plugins/plan_route
[mapviz-1] [INFO] [1608307700.339670910] [mapviz]: Found mapviz plugin: mapviz_plugins/point_click_publisher
[mapviz-1] [INFO] [1608307700.339675828] [mapviz]: Found mapviz plugin: mapviz_plugins/pointcloud2
[mapviz-1] [INFO] [1608307700.339680510] [mapviz]: Found mapviz plugin: mapviz_plugins/pose
[mapviz-1] [INFO] [1608307700.339685234] [mapviz]: Found mapviz plugin: mapviz_plugins/robot_image
[mapviz-1] [INFO] [1608307700.339689846] [mapviz]: Found mapviz plugin: mapviz_plugins/route
[mapviz-1] [INFO] [1608307700.339694707] [mapviz]: Found mapviz plugin: mapviz_plugins/string
[mapviz-1] [INFO] [1608307700.339699451] [mapviz]: Found mapviz plugin: mapviz_plugins/textured_marker
[mapviz-1] [INFO] [1608307700.339704054] [mapviz]: Found mapviz plugin: mapviz_plugins/tf_frame
[mapviz-1] [INFO] [1608307700.339708697] [mapviz]: Found mapviz plugin: mapviz_plugins/tile_map
[... cut ...]
[... 23 lines cut ...]
  Service Clients:

  Action Servers:

  Action Clients:

$ ros2 topic echo /local_xy_origin
1608308001.145427 [230]       ros2: using network interface wlp82s0 (udp/192.168.2.109) selected arbitrarily from: wlp82s0, br-1a95fe46b61d, br-63ba9ba7b1aa, docker0, br-ecc4325848b8, tun0
header:
  stamp:
    sec: 0
    nanosec: 0
  frame_id: map
pose:
  position:
    x: -98.61370577
    y: 29.45196669
    z: 233.719
  orientation:
    x: 0.0
    y: 0.0
    z: 0.0
    w: 1.0
---
```

nicolasaw, 2020-12-19:
Hi, sorry for the late reply due to the time difference. I somehow succeeded in opening the Mapviz as cleanly as you did. 
[image]

The problem I found is that, I was launching with the Python launch file, not with XML launch file as in ROS2 Tutorials they used Python launch file. So I have been launching with `ros2 launch mapviz mapviz.launch.py`. Is there any significance reason why the Python launch file does not work? 

Btw, thanks in advance

closed 2021-01-05 by nicolasaw

kimsooyoung, 2022-01-07:
@nicolasaw 

I reviewed this issue and was truely helpful to me, Thank you.

BTW, **I found some mistakes in python launch file and wanna share about that.**

In order to pass params. I changed some format in python launch file.
 
Here's some example

```python
      launch_ros.actions.Node(
            package="swri_transform_util",
            executable="initialize_origin.py",
            name="initialize_origin",
            parameters=[
                {"local_xy_frame" : "map"},
                {"local_xy_origin" : "auto"},
                {"local_xy_navsatfix_topic" : "/gps/fix"},
            ]
        ),
```  

In the [initialize_origin.py](https://github.com/swri-robotics/marti_common/blob/dashing-devel/swri_transform_util/nodes/initialize_origin.py), you can find default params and how it works.

Hope to have nice day && this helps to somebody 🤞

crystal-butler, 2022-03-30:
@kimsooyoung  is right about the parameters in `mapviz.launch.py`. I had to modify their definitions as described to get the file to run.

It sounds like @nicolasaw got `mapviz.launch.xml` to work, but I get the following error:
```
launch.invalid_launch_file_error.InvalidLaunchFileError: Caught exception when trying to load file of format [xml]: Attribute value of Entity param expected to be of type None.`       [{ name: swri,          latitude: 29.45196669,          longitude: -98.61370577,          altitude: 233.719,          heading: 0.0},                  { name: back_40,          latitude: 29.447507,          longitude: -98.629367,          altitude: 200.0,       heading: 0.0}]` can not be converted to one of those types
```
This is with ROS 2 Galactic. I was able to launch using the XML file under an install of MapViz under ROS 2 Eloquent, so not sure what might be different.

To the great explanation from @pjreed, I want to say that I'm looking to use a static origin in the visualization I'm currently working on, and there is no GPS message available to set the origin. So I need to be able to feed static coordinates into `initialize_origin.py`. At the moment that doesn't seem to be possible with either launch file as they're coded in the `galactic-devel` branch, unless my problem with the XML file is related to my installation or environment somehow.

timdodge54, 2022-08-23:
So, I recently ran into this problem trying to launch mapviz within a larger python ros launch file and could not auto assign local_xy_origin or use the xml launch. The problem with the original launch file is the list of dictionaries is not being passed in as a string the correct format a node would be like this.
```
     Node(
                package="swri_transform_util",
                executable="initialize_origin.py",
                name="initialize_origin",
                # emulate_tty=True,
                # output='screen',
                parameters=[
                    {"local_xy_frame": "slc"},
                    {"local_xy_origin": "slc"},
                    {
                        "local_xy_origins": """[
                        {"name": "slc",
                        "latitude": 40.76935585340519,
                        "longitude":  -111.89730834960938,
                        "altitude": 0.0,
                        "heading": 0.0
                        }
                ]"""
                    },
            ]
    )
```

wellscrosby, 2023-11-19:
I've been having the same issue as everyone else on ROS2 humble, running the XML file results in the same issue as @crystal-butler, and trying the fix @timdodge54 posted results in:
`[initialize_origin.py-2] rclpy.exceptions.InvalidParameterTypeException: Trying to set parameter 'local_xy_origins' to '[
[initialize_origin.py-2]                     {"name": "swri",
[initialize_origin.py-2]                     "latitude": 40.76935585340519,
[initialize_origin.py-2]                     "longitude":  -111.89730834960938,
[initialize_origin.py-2]                     "altitude": 0.0,
[initialize_origin.py-2]                     "heading": 0.0
[initialize_origin.py-2]                     }]' of type 'STRING', expecting type 'DOUBLE_ARRAY'
`
Anyone have any idea what is going on? The local_xy_origin topic has not been set in any attempts I've made.

timdodge54, 2023-11-20:
> I've been having the same issue as everyone else on ROS2 humble, running the XML file results in the same issue as @crystal-butler, and trying the fix @timdodge54 posted results in: [...]

So, it has been awhile since I ran into this issue. But it seems since I posted this there has been a change to how mapviz takes their inputs. It now needs to be an array. Here is my current configuration.

```python
                 Node(
                package="swri_transform_util",
                executable="initialize_origin.py",
                name="initialize_origin",
                parameters=[
                    {"local_xy_frame": "slc"},
                    {"local_xy_origin": "slc"},
                    {
                        "local_xy_origins": [
                            40.76935585340519,
                            -111.89107193491502,
                            0.0,
                            0.0,
                        ]
                    },
                    {"use_sim_time": use_sim_time},
                ],
            )
```

kyuhyong, 2023-12-21:
Hi
I'm struggling with same issue with the latest mapviz source build on ros-foxy.
Keep getting Wgs84Transformer not initialized error.
If you solved the issue, can you share your final launch.py file here?

PR #713 (2020-12-04): Fix launch files in ROS 2 Foxy
There were a bunch of issues in both the XML and Python example launch files that prevented them from working in ROS 2 Foxy.  This fixes them.

Fixes #712

Distribution Statement A; OPSEC #2893