swri-robotics/mapviz #720: Allow building with C++14 for versions > ROS Kinetic

rjb0026, 2021-02-21:
ROS Melodic and greater default builds to target C++14. I think all mapviz packages should default to those settings unless building for a version of ROS <= Kinetic, in which case builds should target C++11.

matt-attack, 2021-03-02:
Is there a particular motivation for this since code will currently have to work with 11 anyways? ABI issues when using exterior libraries?

rjb0026, 2021-03-02:
Just trying to get ahead of this pull request for a dependency (swri_yaml_util) https://github.com/swri-robotics/marti_common/pull/615

Some of the targets in mapviz that depend on that library were not targeting a modern version of c++ and causing a build failure in kinetic when while I was testing that pull request.

PR #721 (2021-03-05): 720 allow building with c++14 for version greater than kinetic
Fixes #720
closed 2021-03-05 by matt-attack