swri-robotics/mapviz #722: NETWORK ERROR: Protocol "" is unknown

tranqkhue, 2021-02-23:
Hello,
My mapviz does show blank tile_map, although status is `OK` . The error is `NETWORK ERROR: Protocol "" is unknown` . My mapviz config is minimal:
```
capture_directory: "~"
fixed_frame: map
target_frame: <none>
fix_orientation: false
rotate_90: false
enable_antialiasing: true
show_displays: true
show_status_bar: true
show_capture_tools: true
window_width: 1287
window_height: 852
view_scale: 0.2709693
offset_x: 0
offset_y: 0
use_latest_transforms: true
background: "#a0a0a4"
displays:
  - type: mapviz_plugins/tile_map
    name: new display
    config:
      visible: true
      collapsed: false
      custom_sources:
        []
      bing_api_key: [hidden]
      source: Bing Maps (terrain)
```
What's this issue and how to resolve it? Thank you :D

lpbonomi, 2023-09-19:
Have the same error

danthony06, 2023-09-19:
Can you tell me which platform and version you are running?

juanimir25, 2023-09-19:
Ubuntu 20.04.6 LTS and ROS noetic

danthony06, 2023-09-19:
This is on an x86 machine and not Arm? Is the DISPLAY variable set in your
terminal?

lpbonomi, 2023-09-20:
We both have x86 machines, when I `echo $DISPLAY` the output is `:0`

danthony06, 2023-09-20:
I talked with some of the people around here, and our current suspect is that the Bing API is not always stable, and we need better error handling and recovery when it throws errors. We're going to try and add this in. It may mean the tile_map isn't always visible because Bing isn't serving the maps correctly, but we'll at least improve the error output so you know that's the problem.

lpbonomi, 2023-09-22:
Thanks a lot @danthony06. We were trying with other options (#791) but no luck. I'm happy to provide any further info that you need

luvwinnie, 2023-10-17:
I'm trying to use this plugin but it doesn't show on the tilemap. Is there any solution?

juanimir25, 2023-10-18:
We use this solution and can use the map on map viz [https://github.com/danielsnider/MapViz-Tile-Map-Google-Maps-Satellite]

agyoungs, 2023-11-20:
This issue was fixed by in https://github.com/swri-robotics/mapviz/pull/800 and can now be closed