swri-robotics/mapviz #724: Can't start Mapviz(process died)

Nomad-Learner, 2021-03-31:
Hi, the problem is mapviz is died on start without gui appear. My ROS is noetic, ubuntu 20.04 - WSL
Is there any need for predefined config file for mapviz, does it make default on start? It seems mapviz can't load, but I'm not sure if this is the only reason.
My command: roslaunch mapviz mapviz.launch

output
process[rosout-1]: started with pid [4802]
started core service [/rosout]
process[mapviz-2]: started with pid [4808]
/opt/ros/noetic/lib/mapviz/mapviz: error while loading shared libraries: libQt5Core.so.5: cannot open shared object file: No such file or directory
process[initialize_origin-3]: started with pid [4810]
process[swri_transform-4]: started with pid [4811]
[mapviz-2] process has died [pid 4808, exit code 127, cmd /opt/ros/noetic/lib/mapviz/mapviz __name:=mapviz __log:=/home/matbi/.ros/log/924327e2-91b5-11eb-9740-2cf05d3f275f/mapviz-2.log].
log file: /home/matbi/.ros/log/924327e2-91b5-11eb-9740-2cf05d3f275f/mapviz-2*.log

Nomad-Learner, 2021-03-31:
Thank you for answer
I solve the problem to use 
sudo strip --remove-section=.note.ABI-tag /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
this command

now I can see mapviz gui

Nomad-Learner, 2021-03-31:
but the other problem is coming

[ERROR] [1617151593.749201500]: "map" passed to lookupTransform argument target_frame does not exist.

I can't understand what the program want

jgassaway, 2021-03-31:
The "map" is a tf frame that Mapviz uses to transform points from one coordinate frame into another. The standard ROS tf frames are described in https://www.ros.org/reps/rep-0105.html. 
In your case, the word "map" is likely the default value in the top left box that says `Fixed Frame`. Mapviz is basically asking for a tf frame with the name "map" to be published (because this is the frame Mapviz is trying to render the data into). If Mapviz is listening to to any data topics (marker arrays, point clouds, odometry messages), you will need to publish tf frames that link the frame id of these incoming messages to the `Fixed Frame` in mapviz. To learn more about tf frames, and publishing them you can look at http://wiki.ros.org/tf/Tutorials.

jgassaway, 2021-03-31:
As a super simple "fix" to just publish a map frame and a fixed odom frame, you could run a static tf publisher:
The args are: static_transform_publisher x y z yaw pitch roll frame_id child_frame_id period_in_ms

$ `rosrun tf static_transform_publisher 0 0 0 0 0 0 map odom 10`

This will publish a tf link with the map frame as a parent, and odom as a child, with no rotation or translation between them, at 10 Hz. Normally, this map to odom frame publishing is handled by some form of odometry measurement (wheel encoders, visual odom) for robotics endeavors, but it might help you get started.

Nomad-Learner, 2021-04-03:
thank you so much answering!
I will try again.

danthony06, 2021-06-11:
@Nomad-Learner I'm curious if you managed to get everything working like you wanted.

Nomad-Learner, 2021-06-11:
> @Nomad-Learner I'm curious if you managed to get everything working like you wanted.

I reinstalled my WSL but I cant handle my problem
so I  installed VMware. In Vmware, mapviz works very good
Now things went well.
Thank you for answering

Nomad-Learner, 2021-06-11:
> @Nomad-Learner I'm curious if you managed to get everything working like you wanted.

> As a super simple "fix" to just publish a map frame and a fixed odom frame, you could run a static tf publisher:
> The args are: static_transform_publisher x y z yaw pitch roll frame_id child_frame_id [...]
ddd

Nomad-Learner, 2021-06-11:
ddd

closed 2021-06-11 by danthony06