swri-robotics/mapviz #726: Can't start Mapviz(process died)-2 another problem

Nomad-Learner, 2021-04-06:
Hi, the problem is mapviz is died on start without gui appear. My ROS is melodic , ubuntu 18.04 - WSL
Is there any need for predefined config file for mapviz, does it make default on start? It seems mapviz can't load, but I'm not sure if this is the only reason.
My command: roslaunch mapviz mapviz.launch

output
[mapviz-2] process has died [pid 28355, exit code 1, cmd /opt/ros/melodic/lib/mapviz/mapviz __name:=mapviz __log:=/home/rosdevelop/.ros/log/5fb67bd6-96d7-11eb-a30d-00155dcb3a95/mapviz-2.log].
log file: /home/rosdevelop/.ros/log/5fb67bd6-96d7-11eb-a30d-00155dcb3a95/mapviz-2*.log

Nomad-Learner, 2021-04-06:
I just changed my version. and the other problem is comming...

I dont know what is wrong

Nomad-Learner, 2021-04-06:
" ============================================================================
" Netrw Directory Listing                                        (netrw v156)
"   /home/rosdevelop/.ros/log/879ff5f8-96d9-11eb-a6ea-00155dcb3a95
"   Sorted by      name
"   Sort sequence: [\/]$,\<core\%(\.\d\+\)\=\>,\.h$,\.c$,\.cpp$,\~\=\*$,*,\.o$,\.obj$,\.info$,\.swp$,\.bak$,\~$
"   Quick Help: <F1>:help  -:go up dir  D:delete  R:rename  s:sort-by  x:special
" ==============================================================================
../
./
initialize_origin-3-stdout.log
initialize_origin-3.log
mapviz-2-stdout.log
master.log
roslaunch-DESKTOP-~~~~-!!!!!.log
rosout-1-stdout.log
rosout.log
swri_transform-4-stdout.log

here is log

Nomad-Learner, 2021-04-06:
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-rosdevelop'
qt.qpa.screen: QXcbConnection: Could not connect to display
Could not connect to any X display.

I think this is problem but how can I solve this problem?

danthony06, 2021-04-06:
It looks like you're still having issues with launching graphical
applications from WSL. Have you tried running other graphical programs in
your system? Are you able to run RViz, for instance?

Nomad-Learner, 2021-04-06:
wow you are right
My command : rosrun rviz rviz
output
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-rosdevelop'
qt.qpa.screen: QXcbConnection: Could not connect to display
Could not connect to any X display.

the same problem is occur

Nomad-Learner, 2021-04-06:
> It looks like you're still having issues with launching graphical applications from WSL. Have you tried running other graphical programs in your system? Are you able to run RViz, for instance?
> […](#)

wow you are right
My command : rosrun rviz rviz
output
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-rosdevelop'
qt.qpa.screen: QXcbConnection: Could not connect to display
Could not connect to any X display.

the same problem is occur

pjreed, 2021-04-06:
I've never used WSL, but I'm going to guess that your problem is because you don't have an X server running.  You might try going through this walkthrough and see if it works for you: https://janbernloehr.de/2017/06/10/ros-windows

Nomad-Learner, 2021-04-06:
> I've never used WSL, but I'm going to guess that your problem is because you don't have an X server running. You might try going through this walkthrough and see if it works for you: https://janbernloehr.de/2017/06/10/ros-windows

thank you for answering but it didn't work 

output
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-rosdevelop'
qt.qpa.screen: QXcbConnection: Could not connect to display :0
Could not connect to any X display.
[mapviz-2] process has died [pid 182, exit code 1, cmd /opt/ros/melodic/lib/mapviz/mapviz __name:=mapviz __log:=/home/rosdevelop/.ros/log/26a00e20-96f5-11eb-8228-00155d229c6f/mapviz-2.log].
log file: /home/rosdevelop/.ros/log/26a00e20-96f5-11eb-8228-00155d229c6f/mapviz-2*.log

Nomad-Learner, 2021-04-06:
Now I have only one problem 

started core service [/rosout]
process[mapviz-2]: started with pid [8229]
process[initialize_origin-3]: started with pid [8230]
process[swri_transform-4]: started with pid [8231]
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-rosdevelop'

how can I take this problem? 

XDG_RUNTIME_DIR not set

Nomad-Learner, 2021-04-06:
> I've never used WSL, but I'm going to guess that your problem is because you don't have an X server running. You might try going through this walkthrough and see if it works for you: https://janbernloehr.de/2017/06/10/ros-windows

something work!

closed 2021-06-11 by danthony06