swri-robotics/mapviz #727: how can I take  XDG_RUNTIME_DIR not set problem

Nomad-Learner, 2021-04-06:
thanks to many people

there is only one left problem. 

process[rosout-1]: started with pid [10369]
started core service [/rosout]
process[mapviz-2]: started with pid [10373]
process[initialize_origin-3]: started with pid [10377]
process[swri_transform-4]: started with pid [10378]
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-rosdevelop'

I serached  XDG_RUNTIME_DIR not set again and again

but I cant find solution. what should I do to solve that problem?

danthony06, 2021-04-07:
It looks like XDG_RUNTIME_DIR is falling back to its default value, and
that shouldn't be a problem any more. I don't have WSL, and I'm not sure if
any of the contributors use it either.

Nomad-Learner, 2021-04-07:
> It looks like XDG_RUNTIME_DIR is falling back to its default value, and that shouldn't be a problem any more. I don't have WSL, and I'm not sure if any of the contributors use it either.
> […](#)

thank you for answering I solve the problem to use this link
https://stackoverflow.com/questions/59790350/qstandardpaths-xdg-runtime-dir-not-set-defaulting-to-tmp-runtime-aadithyasb

hmm..

 but program is stock

SUMMARY
========

PARAMETERS
 * /initialize_origin/local_xy_frame: /map
 * /initialize_origin/local_xy_origin: swri
 * /initialize_origin/local_xy_origins: [{'latitude': 29....
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    initialize_origin (swri_transform_util/initialize_origin.py)
    mapviz (mapviz/mapviz)
    swri_transform (tf/static_transform_publisher)

ROS_MASTER_URI=http://localhost:11311

process[mapviz-1]: started with pid [2501]
process[initialize_origin-2]: started with pid [2502]
process[swri_transform-3]: started with pid [2503]

I wait may be 15 minute but nothing change

Nomad-Learner, 2021-04-07:
> It looks like XDG_RUNTIME_DIR is falling back to its default value, and that shouldn't be a problem any more. I don't have WSL, and I'm not sure if any of the contributors use it either.
> […](#)

thank you for answering
I think WSL2 has problem
qt.qpa.screen: QXcbConnection: Could not connect to display
maybe qt is not familiar with WSL2

danthony06, 2021-04-23:
I'm kind of curious to see if this update from Microsoft fixes your problem:

https://devblogs.microsoft.com/commandline/the-initial-preview-of-gui-app-support-is-now-available-for-the-windows-subsystem-for-linux-2/

If that doesn't fix it, I will probably close the issue. If you find a way to fix the problem, I'd be happy to add it to the documentation or merge in code fixes that address the problem.

Nomad-Learner, 2021-04-24:
Thank you for answering
I will check that document
anyway I just install VMware, And reinstall mapviz
In VMware, mapviz works well

closed 2021-06-11 by danthony06