swri-robotics/mapviz #728: Real time navigation with Mapviz ROS2

nicolasaw, 2021-04-10:
Hi everyone, I am quite new on Mapviz and I am currently using it for my research project. I would like to utilize Mapviz to visualize the movement of my robot equipped with a GPS on a map. Not only that, I also wish to be able to navigate my robot from initial point to a goal point directly on Mapviz as what RViz is capable of. 

https://www.oreilly.com/library/view/ros-programming-building/9781788627436/assets/fb328a98-7d69-4888-8624-4b00029d8365.png

I read in #507 that for Mapviz on ROS1, move_base and plan_route plugins can be used to help navigating. For ROS2, is there something similar to it and how to actually use it? It would be great if it publishes `cmd_vel` topic as that is the topic I will be using to send data to my microcontroller. 

Attach below is the output I was able to get currently, sorry for the lack of quality. 
[image]

Thanks and I will be looking forward to any guidance

pjreed, 2021-04-12:
The way actions work in ROS2 have some significant changes from ROS1, and at the time I originally ported mapviz to ROS2, [MoveIt](https://github.com/ros-planning/moveit2) was still under heavy development and changing rapidly, so I decided not to port the `move_base` plugin, and I believe it's still the one plugin in the ROS1 version of mapviz that hasn't been ported to ROS2.  Unfortunately, I still not familiar with ROS2 actions or MoveIt, but anybody who wants to take a look at porting that plugin would be welcome to do so. 😉

kyuhyong, 2023-12-20:
Hi @nicolasaw 
How were you able to show tile_map on mapviz?
I source build mapviz on my ros2_ws but I got Wgs84Transformer not initialized error and no map shows.
Can you share your launch.py for a reference?