swri-robotics/mapviz #729: [initialize_origin-2] process has died

ThornsOoO, 2021-04-14:
Hi, the problem is mapviz is died. My ROS is melodic, ubuntu 18.04
It's strange that it can still run in the morning, but it can't load tile_map in the afternoon
my command: roslaunch mapviz mapviz.launch

output
ImportError: /home/cyk/catkin_ws/devel/lib/python3/dist-packages/tf2_py/_tf2.so: undefined symbol: PyModule_Create2
[initialize_origin-2] process has died [pid 9938, exit code 1, cmd /home/cyk/catkin_ws/src/marti_common/swri_transform_util/nodes/initialize_origin.py fix:=/fix __name:=initialize_origin __log:=/home/cyk/.ros/log/5af053d0-9d01-11eb-bc6a-70b5e8388aac/initialize_origin-2.log].
log file: /home/cyk/.ros/log/5af053d0-9d01-11eb-bc6a-70b5e8388aac/initialize_origin-2*.log

Thanks and I will be looking forward to any guidance

launch :
<launch>

  <node pkg="mapviz" type="mapviz" name="mapviz"></node>

  <node pkg="swri_transform_util" type="initialize_origin.py" name="initialize_origin" >
    <param name="local_xy_frame" value="/map"/>
    <param name="local_xy_origin" value="auto"/>
    <rosparam param="local_xy_origins">
      [{ name: swri,
         latitude: 29.45196669,
         longitude: -98.61370577,
         altitude: 233.719,
         heading: 0.0},
         
       { name: back_40,
         latitude: 29.447507,
         longitude: -98.629367,
         altitude: 200.0,
         heading: 0.0}]
    </rosparam>
    <remap from="fix" to="/fix"/>
  </node>

  <node pkg="tf" type="static_transform_publisher" name="swri_transform" args="0 0 0 0 0 0 /map /origin 100"  />

</launch>

closed 2021-04-14 by ThornsOoO

pjreed, 2021-04-14:
Since you closed this, I'm going to assume you figured it out, but just in case...

This stands out as being very suspicious to me:
```
/home/cyk/catkin_ws/devel/lib/python3/dist-packages/tf2_py/_tf2.so
```

ROS Melodic does not support Python 3; this line makes me think that maybe you're trying to make a custom build of the `tf2_py` package that is using Python 3.  The problem here is somewhere deeper than the initialize_origin node; I'm guessing that somewhere in your environment, you have a mix of Python 2 & 3 libraries, and it's crashing when a single process tries to load both of them at once.

Probably the easiest solution is to either run everything in Python 2 or upgrade to Ubuntu 20.04 / ROS Noetic if you need Python 3 support.

ThornsOoO, 2021-04-14:
First of all, thank you very much for your reply。I closed because I found my problem is very stupid 。I solved it through  source. zshrc.   Maybe it's because a python extension package was installed at noon
Anyway, thank you again for your attention and reply

 

 
ROS Melodic does not support Python 3; this line makes me think that maybe you're trying to make a custom build of the tf2_py package that is using Python 3.  The problem here is somewhere deeper than the initialize_origin node; I'm guessing that somewhere in your environment, you have a mix of Python 2 &amp; 3 libraries, and it's crashing when a single process tries to load both of them at once.

ThornsOoO, 2021-04-14:
The problem of python2 and python3 is really annoying, but some packages rely on python2. And I am a beginner,so I dare not  to change the version at will,  I'm afraid it will affect other packages.. This is my first time to submit a question on GitHub.I am very moved by your reply.Thank you very much for your answer

pjreed, 2021-04-14:
No problem, feel free to let us know if you have any other issues.