swri-robotics/mapviz #731: [Foxy] multires_image doesn't export dependencies

jonselling, 2021-05-24:
Looking at the CMakeLists.txt for multires_image a lot of packages are pulled in but the dependencies are exported are in a non initialized variable "RUNTIME_DEPS", and then Qt.

Is that variable filled elsewhere or we should update the RUNTIME_DEPS to include the packages that it finds? Also, It seems exporting "Qt" is not correct, should it at last be "Qt5" or the specific Qt5 packages that are there?

On a separate note, if it is still a problem in "swri_transform_util" with OpenCV where if you use components you should declare them in a CONFIG_EXTRAS: https://answers.ros.org/question/331089/ament_export_dependenciesboost-not-working/?answer=332460#post-id-332460. That can be a ticket created in the right repo though, just saw the comment when pulling in OpenCV here, but it might apply here since it looks like we are depending on components of Boost.

vknisley-swri, 2021-08-31:
I'm looking into the Qt export issue now.

danthony06, 2021-09-01:
Resolved via https://github.com/swri-robotics/mapviz/pull/742

PR #742 (2021-09-01): 731 Fix exported dependencies in multires_image (Foxy / dashing-devel)
I set the `RUNTIME_DEPS` variable and included the different, specific Qt5 packages within it. I also added a `CONFIG_EXTRAS` file for the Boost components. Please let me know if I need to make any further changes.

Closes #731