swri-robotics/mapviz #739: Marker Scale

tky1026, 2021-07-26:
The `visualization_msgs::Marker` do not scale according to the size provided when showing up in MapViz's marker(display).

For example when displaying `Marker` type

```
marker.type = visualization_msgs::Marker::POINTS;
marker.scale.x = 1.0;
marker.scale.y = 1.0;
```
Instead of showing up according the scale (1.0m) similar to RViz, the are shown as one single pixel only in MapViz.

[image]

[image]

It would be nice, if the markers can show according to size, or maybe there are some ways to do it that i missed.

Thank you.

danthony06, 2021-07-26:
@matt-attack do you have any insight into this?

pjreed, 2021-07-26:
IIRC, this has been a known issue for a long time.  The `scale` in a `visualization_msgs/Marker` message is supposed to be in meters, which is how they behave in rviz, but mapviz interprets it as pixels.

The obvious solution to "fix" things here is to make Mapviz draw them in meters instead, but the problem is that there are many places where nodes that were developed and tested primarily with Mapviz publish markers and expect the scale to be in pixels, and so changing that will cause all of those old markers to be way too large (or small, depending on how zoomed out you are).

For now I would suggest adding a button to the Marker config panel that toggles whether scales are in meters or pixels and have it default to pixels to preserve backwards compatibility, but in the future it should probably be in meters instead.