swri-robotics/mapviz #740: TEXT_VIEW_FACING markers don't disappear when namespace is hidden

DangitBen, 2021-08-11:
### Info
Mapviz Version: 1.4.1-1bionic.20210716.100718

### Terminal 1
```bash
rostopic pub /markers visualization_msgs/MarkerArray \
"markers: 
  - header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs: 0
      frame_id: 'map'
    ns: 'text'
    id: 0
    type: 9
    action: 0
    pose: 
      position: 
        x: 5.0
        y: 5.0
        z: 0.0
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 0.0
    scale: 
      x: 0.75
      y: 0.0
      z: 0.0
    color: 
      r: 0.75
      g: 0.0
      b: 1.0
      a: 1.0
    lifetime: 
      secs: 0
      nsecs:         0
    frame_locked: False
    points: []
    colors: []
    text: 'you cant hide me! :P'
    mesh_resource: ''
    mesh_use_embedded_materials: false"
```

### Terminal 2
```bash
rosrun mapviz mapviz
```

### Instructions
- Clear mapviz config
- Add a marker plugin, and select the above topic.
- Uncheck the marker namespace `text` and see that the marker doesn't disappear 

This could also happen with other marker types, but this functionality **does** work with LINE_STRIP marker types.

matt-attack, 2021-08-12:
Does this work when you build mapviz from source? We relatively recently added a patch for specifically this here: https://github.com/swri-robotics/mapviz/blob/master/mapviz_plugins/src/marker_plugin.cpp#L636

DangitBen, 2021-08-13:
It does work when built from source! I'll close the issue.

closed 2021-08-13 by DangitBen

matt-attack, 2021-08-13:
We'll try and get a new release pushed out once we figure out the new process.