swri-robotics/mapviz #741: Publish clicked point with orientation

tky1026, 2021-08-24:
I would like to publish GPS coordinate message by clicking on the display, where `point_click_publisher` with frame set to `/wgs84` get the job done perfectly.

However, I am looking to add orientation like how the move_base 2D Estimated Pose option works, such that the message field contains GPS coordinate + orientation.

`point_click_publisher` publishes `PointStamped` only;
`move_base Estimated Pose` publishes pose message to `/initialpose` with data under map frame only and not changeable.

How can I achieved this?

matt-attack, 2021-08-24:
There is not an existing way to do this using just mapviz.

One way we have done something like this is by creating a node which listens to two PointStamps and uses the relative heading between the two to create a Pose. The alternative is creating a new mapviz plugin based on point_click_publisher (or adding an option to it) which does something similar internally.

danthony06, 2021-08-24:
I'm thinking about how we could integrate this more easily into the UI. What if we did something like a click and drag, where you click where the waypoint should be, and then drag to the direction you want? After you release, the point would be published.

knickels, 2021-08-24:
@tky1026, check out https://github.com/swri-robotics/mapviz/pull/716 and see if it does what you need.  I based it off of the move_base plugin.