swri-robotics/mapviz #743: mapviz crashes in raspi 4 running on Ubuntu 20.04

siddharthcb, 2021-09-16:
Thanks for this package.
my build was successful on raspi 4b booted with ubuntu 20.04. however, I am unable to get the gui there is no particular error displayed on the terminal either.

I have tried building from source and apt-installing it but both gives me the same error.

```
$ roslaunch mapviz mapviz.launch

... logging to /home/ubuntu/.ros/log/82da555e-16d6-11ec-81b5-5f230f1d19e1/roslaunch-ubuntu-44647.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/ubuntu/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://192.168.65.111:46311/

SUMMARY
========

PARAMETERS
 * /initialize_origin/local_xy_frame: /map
 * /initialize_origin/local_xy_origin: swri
 * /initialize_origin/local_xy_origins: [{'name': 'swri',...
 * /rosdistro: noetic
 * /rosversion: 1.15.11

NODES
  /
    initialize_origin (swri_transform_util/initialize_origin.py)
    mapviz (mapviz/mapviz)
    swri_transform (tf/static_transform_publisher)

auto-starting new master
process[master]: started with pid [44657]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 82da555e-16d6-11ec-81b5-5f230f1d19e1
process[rosout-1]: started with pid [44667]
started core service [/rosout]
process[mapviz-2]: started with pid [44670]
process[initialize_origin-3]: started with pid [44675]
process[swri_transform-4]: started with pid [44676]
[ERROR] [1631787116.865584429]: Failed to load file: /home/ubuntu/.mapviz_config
[mapviz-2] process has died [pid 44670, exit code -11, cmd /home/ubuntu/catkin_ws/devel/lib/mapviz/mapviz __name:=mapviz __log:=/home/ubuntu/.ros/log/82da555e-16d6-11ec-81b5-5f230f1d19e1/mapviz-2.log].
log file: /home/ubuntu/.ros/log/82da555e-16d6-11ec-81b5-5f230f1d19e1/mapviz-2*.log
```

```
mapviz.launch file:

<launch>

  <node pkg="mapviz" type="mapviz" name="mapviz"></node>

  <node pkg="swri_transform_util" type="initialize_origin.py" name="initialize_origin" >
    <param name="local_xy_frame" value="/map"/>
    <param name="local_xy_origin" value="swri"/>
    <rosparam param="local_xy_origins">
      [{ name: swri,
         latitude: 29.45196669,
         longitude: -98.61370577,
         altitude: 233.719,
         heading: 0.0},

       { name: back_40,
         latitude: 29.447507,
         longitude: -98.629367,
         altitude: 200.0,
         heading: 0.0}]
    </rosparam>
  </node>

  <node pkg="tf" type="static_transform_publisher" name="swri_transform" args="0 0 0 0 0 0 /map /origin 100"  />

</launch>

```
`sudo apt update && sudo apt upgrade` did not solve the problem either.
Any insights on this would be helpful.

danthony06, 2021-09-16:
I suspect it's something similar to the issue here: https://github.com/swri-robotics/mapviz/issues/660. I'm not sure if 20.04 made fixing this any easier, but we could take a look at it again.

DrawingProcess, 2023-07-12:
I got a same problem. Is there any solution?

danthony06, 2023-07-12:
Mapviz is unsupported for Raspberry Pis due to their OpenGL implementation, see here for further discussion: https://github.com/swri-robotics/mapviz/issues/580. Fixing this would require re-writing the backend rendering, and we are not planning on do so at this time.

closed 2023-07-12 by danthony06