swri-robotics/mapviz #747: Cannot load map +  Wgs84Transformer not initialized + NETWORK ERROR

robodyne-robotics, 2021-11-20:
I'm using ROS Melodic on Ubuntu 18.04 to launch mapviz.
I'm not able to load any map (i.e. http://maps.stamen.com/terrain/#12/40.4485/17.9240) since as soon as I load the tilemap plugin I get:
```
[ WARN] [1637403046.945206940]: Wgs84Transformer not initialized
[ERROR] [1637403046.945483290]: Error: No transform between /wgs84 and map
[ WARN] [1637403048.950361959]: Wgs84Transformer not initialized
```

How can I fix this problem?
This is my launch file:

```
<launch>

  <node pkg="mapviz" type="mapviz" name="mapviz"></node>

  <node pkg="swri_transform_util" type="initialize_origin.py" name="initialize_origin" >
    <param name="local_xy_frame" value="/map"/>
    <param name="local_xy_origin" value="auto"/>
    <!--<param name="local_xy_origin" value="swri"/>-->
    <rosparam param="local_xy_origins">
      [{ name: swri,
         latitude: 40.5675262,
         longitude: 17.4155641,
         altitude: 133.719,
         heading: 0.0},

       { name: back_40,
         latitude: 40.5675262,
         longitude: 17.4155641,
         altitude: 100.0,
         heading: 0.0}]
    </rosparam>
    <remap from="fix" to="/gnss"/>
  </node>

  <node pkg="tf" type="static_transform_publisher" name="swri_transform" args="0 0 0 0 0 0 /map /origin 100"  />

</launch>
```

danthony06, 2021-12-03:
I notice the local_xy_origin param is called local_xy_origins. Could that be the problem?

robodyne-robotics, 2021-12-03:
Thank you for your answer and support! How should I change it in order to fix the issue?
I don't think I have changed it.

danthony06, 2021-12-03:
I suspect you might be running into something similar to this: https://github.com/swri-robotics/mapviz/issues/712#issuecomment-748182040. Can you echo ```/local_xy_origin``` to see if you the origin is getting published?

lpbonomi, 2023-09-17:
I'm getting the same error.

I was setting up mapviz following this [tutorial](https://medium.com/@hitlx916/visualize-the-gnss-messages-in-mapviz-ros-4ae7eec19936) and after clicking the selector for `Color` it freezed. Later, when trying to restart the app I got this errors:

```
[ WARN] [1694950799.304974569]: Wgs84Transformer not initialized
[ERROR] [1694950799.305586765]: Error: No transform between /wgs84 and map
[ WARN] [1694950799.590368388]: No messages received.

(mapviz:10376): GLib-GObject-WARNING **: 08:39:59.709: cannot register existing type 'GtkStyle'

(mapviz:10376): GLib-GObject-WARNING **: 08:39:59.709: cannot add private field to invalid (non-instantiatable) type '<invalid>'

(mapviz:10376): GLib-CRITICAL **: 08:39:59.709: g_once_init_leave: assertion 'result != 0' failed

(mapviz:10376): Gtk-CRITICAL **: 08:39:59.709: gtk_style_attach: assertion 'GTK_IS_STYLE (style)' failed
[mapviz-1] process has died [pid 10376, exit code -11, cmd /opt/ros/noetic/lib/mapviz/mapviz __name:=mapviz __log:=/home/odroid/.ros/log/879aa2e8-554e-11ee-9d03-ce660a7141e3/mapviz-1.log].
log file: /home/odroid/.ros/log/879aa2e8-554e-11ee-9d03-ce660a7141e3/mapviz-1*.log
```

lpbonomi, 2023-09-17:
Could fix this crashes by removing the config of mapviz.

You can find the config by running `sudo find / -name "*.mapviz_config"`