swri-robotics/mapviz #768: Pre-config for robot image plugin

byu-awesome-rover, 2022-11-02:
Hi all,
I am trying to preset a robot image using a config file.
And doing it like below does not load the image at the beginning.
What is the way to load the `image_file`?

```
  - type: mapviz_plugins/robot_image
    name: Rover Image
    config:
      visible: true
      collapsed: false
      image_file: ~/BYU-Mars-Rover/rover_ws/src/mapviz_tf/Turtle.png
      frame: origin
      width: 50.00
      height: 50.00
      offset_x: 0
      offset_y: 0
```

malban, 2022-11-02:
It looks like the robot_image plugin doesn't know what `~/` means and is expecting either an absolute path or in the following form for a package relative path: `$(find mapviz_tf)/Turtle.png`

pjreed, 2022-11-02:
By the way, using `~` to represent your home directory is a `sh`/`bash`-specific substition, so that generally won't work in anything that isn't using the shell to resolve paths.

`robot_image_plugin.cpp` has some code specifically to handle using `$(find ...)` to look up a ROS package path, but it doesn't handle any other substitutions.  If you need to be able to resolve things relative to your home directory, it'd probably be very straightforward to also add support for using `$(env ...)` to resolve environment variables; then you could use `$(env HOME)`.

closed 2022-11-02 by danthony06