swri-robotics/mapviz #769: Custom Plugins

aksk1119, 2022-12-05:
Hi all,
Is there a way to add a custom plugin to Mapviz?

Thank you

malban, 2022-12-05:
Yes,  you can look at the [mapviz_plugins](https://github.com/swri-robotics/mapviz/tree/master/mapviz_plugins), [multires_image](https://github.com/swri-robotics/mapviz/tree/master/multires_image), and [tile_map](https://github.com/swri-robotics/mapviz/tree/master/tile_map) packages as examples.

You'll need a [mapviz_plugins.xml](https://github.com/swri-robotics/mapviz/blob/master/mapviz_plugins/mapviz_plugins.xml) file and to reference it in [package.xml](https://github.com/swri-robotics/mapviz/blob/master/mapviz_plugins/package.xml#L49) so mapviz can find the plugin(s).

Basically, you will inherit from the `mapviz::MapvizPlugin` base class and implement the custom functionality.  For example: https://github.com/swri-robotics/mapviz/blob/master/mapviz_plugins/include/mapviz_plugins/grid_plugin.h

aksk1119, 2022-12-05:
Thank you, Marc. I will take a look at them!

closed 2022-12-05 by danthony06