swri-robotics/mapviz #777: [Noetic] mapviz fails to build on armhf

sloretz, 2023-05-23:
Hey @danthony06,

It looks like `mapviz` isn't building on some platforms:

* [Ubuntu Focal armhf](https://build.ros.org/view/Nbin_ufhf_uFhf/job/Nbin_ufhf_uFhf__mapviz__ubuntu_focal_armhf__binary/)

I'm planning to add all chronically failing packages to per-platform blacklists later this week.
That just means the buildfarm won't try to build them every day.

When all issues causing the build failures are resolved, please make a PR to the `production` branch on [ros_buildfarm_config](https://github.com/ros-infrastructure/ros_buildfarm_config/tree/production/noetic) removing it from the `package_blacklist`:

* [release-armhf-build.yaml](https://github.com/ros-infrastructure/ros_buildfarm_config/blob/production/noetic/release-armhf-build.yaml)

danthony06, 2023-05-24:
We shouldn't be trying it at all. I'll try to remove it on our end when I'm back in the office.

closed 2023-08-24 by danthony06

danthony06, 2023-08-24:
@sloretz is there a similar configuration for ROS2?

danthony06, 2023-08-24:
Never mind, found it. Sorry, couldn't find it for a while. I'm going to blacklist this for the ROS2 Arm builds.