swri-robotics/mapviz #784: Release binaries for ROS 2?

SteveMacenski, 2023-08-18:
There is some discussion about using this w.r.t. a tutorial in Nav2 for GPS waypoint following. A blocker for that is that this isn't released in binary format - is there a reason for that and/or is this something that can be released?

Thanks!
S

danthony06, 2023-08-18:
Let me figure it out. I'm happy to make changes or add features to support Nav2 integration, so please let me know if there's anything else we can do as well.

danthony06, 2023-08-18:
@SteveMacenski I've got the process rolling. The binary releases were disabled a while ago due to a dependency issue. I resolved that issue a few months ago, but the builds never got unblocked. I've updated the issue to draw attention to it, and hopefully everything gets rapidly released now: https://github.com/ros2-gbp/mapviz-release/issues/2.

SteveMacenski, 2023-08-18:
Awesome! This sounds like a way to go then!

danthony06, 2023-08-21:
@SteveMacenski it should be unblocked on the buildfarm and available as soon as that process completes.

danthony06, 2023-08-21:
Humble and Iron. If there's a need for Rolling, I can do that as well.

SteveMacenski, 2023-08-21:
Rolling as well would be great since alot of our users are on rolling to get the bleeding edge of Nav2

danthony06, 2023-08-21:
Alright, it's been a while since I've tried on Rolling. I'll try to get all of the dependencies and the package released tonight.

danthony06, 2023-08-22:
Okay, I have it building locally on Rolling and am working on blooming it. Several of the dependencies we maintain haven't been bloomed either, so that's also getting updated.

danthony06, 2023-08-22:
I'm also going to make a big push to get this working on Arm devices. It's been a common request, and I think I know what we need to do to get it going.

SteveMacenski, 2023-08-22:
What deps are missing?

danthony06, 2023-08-22:
marti_common and marti_messages, which are things we maintain. I had pretty
old versions in Rolling, and I just bloomed new versions of them.

danthony06, 2023-08-24:
Okay, I just bloomed Mapviz for rolling. The PR request is here: https://github.com/ros/rosdistro/pull/38343.

closed 2023-08-24 by danthony06

SteveMacenski, 2023-08-25:
Thanks! Tutorial coming soon!

jonselling, 2023-09-08:
It looks like on 22.04 I only see apt packages for `mapviz-interfaces` and not the binary release of mapviz (on humble, iron, or rolling).

Does the `swri_roscpp` builds need to be enabled again? (I believe [this is the rolling build](https://build.ros2.org/job/Rbin_uJ64__swri_roscpp__ubuntu_jammy_amd64__binary/)?)

danthony06, 2023-09-12:
marti_common wasn't building correctly, which was blocking Mapviz. I just
released v3.6.1 of marti_common to the buildfarm, and it looks like it's
building, which should unblock Mapviz on x86 platforms. I already got
marti_common packages unblacklisted on the buildfarm, so everything should
be good now.