swri-robotics/mapviz #790: String message's configuration is  reset when re launch mapviz

ccsmm78, 2023-09-22:
In ubuntu22 humble, branch ros2-devel,
string configuration is not applied to mapviz when it re-starts.
For example, in ~/.mapviz_config the following are not applied to mapviz, but it looks like just default Hevetica-10 size, top left. 

  - type: mapviz_plugins/string
    name: new display
    config:
      visible: true
      collapsed: true
      font: Sans Serif,26,-1,5,50,0,0,0,0,0
      color: "#55aaff"
      topic: /arceye/vl_status
      anchor: bottom center
      units: pixels
      offset_x: 0
      offset_y: 0

danthony06, 2023-09-22:
I'm having troubles reproducing this. Does the string's font change if you change it through the UI?

danthony06, 2023-09-22:
Okay, if I copy in your config I see the same behavior, so I think it's something was saved in a bad state, and the program is silently falling back to its defaults. I'll try to figure out what's wrong with the format, and also improve the behavior when the system fails to load the setting correctly.

Do you have any ideas on how it might have gotten into a bad state? Did the program crash before this?

ccsmm78, 2023-09-26:
Thank you for your kind reply^^.

There was no program crash, and I am using mapviz on several versions of Ubuntu (docker). The default font works fine,  changing its size and position also works well. I've checked that the configuration is well saved in ~/.mapviz_config.

However, when I run the program again, the string is displayed in the default font and position (0,0) instead of the changed settings. This means that the saved configuration of /.mapviz_config for string is not reflected. This problem only occurs with humble-ubuntu22 (package with "git clone -b ros2-devel _here_"). This problem does not occur in melodic-ubuntu18. Apart from this problem, foxy-ubuntu20 has not been tested because the UI update is unstable. 
Of course, other settings except for string (e.g. navsat, pose, etc.) are reflected well.

danthony06, 2023-09-26:
I think it might be related to the '5' in the font description string. I believe it's the font style hint as described here: https://doc.qt.io/qt-5/qfont.html#StyleHint-enum. When I try to set Sans Serif on my test computer, it sets that value to 0. I'm not sure why that would cause a problem.

Could you try changing the 5 in that string to 0 in your configuration and see if that fixes your problem? It would make me more confident that I'm fixing the right problem.

danthony06, 2023-09-26:
I have pushed a change to the `ros2-devel` branch that will at least show us an error when it can't load the correct font. Would you mind giving it a try and telling me if it helps with your problem?