swri-robotics/mapviz #795: Plot Google Maps in RVIZ

Cram3r95, 2023-10-06:
Hi,

Is it possible to plot a Google Maps / OSM layout in RVIZ? I must represent my 3D sensors and markers, so I cannot use mapviz in 2D.

Best,

Carlos

closed 2023-10-06 by danthony06

Cram3r95, 2023-10-06:
@danthony06 Thank you, but I want to represent my data (LiDAR markers, etc.) in the 3D space, and Mapviz is only designed for the 2D space. Is it possible to implement the map as a layout in RVIZ?

danthony06, 2023-10-06:
Mapviz isn't an RViz plugin, and thus can't be embedded with RViz. I'm not aware of any mechanism for showing Google Maps data in RViz, but I also haven't looked into that in a long time. You might check out Foxglove Studio, it might be closer to what you want.