swri-robotics/mapviz #796: Unable to see fix messages

miguelao33, 2023-10-13:
Hello everyone, thanks a lot for mapviz!

I have an ardusimpe ZED-F9P with this node ([https://github.com/aussierobots/ublox_dgnss]) publishing on /fix topic, which I believe is enough at least to see single points of location on the map. 

As you can see, the frame id of the message is "ubx" and I dont know if it has any influence on that. 

[image]

I'm still not running robot_localization_nav_sat_node. tons of covariance (I'm inside the building!)

Any idea of how to represent that data there?

Thanks!

danthony06, 2023-10-13:
Is it an RTK receiver?

miguelao33, 2023-10-13:
No, is configured right now as a Rover. I'll buy another ardusimple board configured as RTK soon.