swri-robotics/mapviz #816: [ROS2] Add ability to change QoS settings for plugin subscribers

rjb0026, 2024-03-29:
It would be nice to have a standard widget that many plugins could use to modify subscriber QoS settings. 2 examples of things we currently can't do with mapviz:
- Ensure that plugins subscribing to a topic with offered durability "transient local" will always get a message when mapviz starts up. This is particularly confusing for publishers that emulate the ROS1 "latched" topic concept and might only publish 1 message on startup. If mapviz starts after this subscriber publishes its single message, the plugin will never receive a message.
- Ensuring that plugins subscribing to topics with offered reliability "best effort" can receive any messages. Currently mapviz plugin subscribers are only able to subscribe with reliability "reliable" which prevents them from making connections with publishers that offer the reliability QoS "best effort", severely limiting the usability of mapviz in highly tuned ROS2 systems.

danthony06, 2024-04-01:
To summarize a conversation we had offline, it would be nice to solve this in such a way that the capability is added to all existing plugins without needing code changes. It would also be nice if this is implemented in such a way that the plugin writers don't have to explicitly add support for it, because that can lead to bugs or plugins with different sets of capabilities.

- Add QoS settings to the dropdown list of topic properties
- When QoS settings are changed, delete and re-create the publisher or subscriber to match the new QoS profile
- Figure out how to save these settings in the YAML configuration file.

rjb0026, 2024-08-11:
closed by #818 and #820

closed 2024-08-11 by rjb0026