swri-robotics/mapviz #831: Retrieve Zoom level via ROS Topics or services.

joeldevasia, 2024-11-19:
Hello there, I am currently using MapViz in ROS Noetic. I noticed that Visualisation Markers don't scale based on zoom. In my particular use case, the user would have to zoom in and out  from few metres of screen area to several hundreds of Kilometres frequently. And the markers I publish are dynamic. So I was wondering if there is no built-in way to resize Visualisation Markers based on zoom, then if some how i could get the zoom level, I would re-publish all markers with updated scale. I could then set a fixed frequency for updating the marker or update them when zoom changes.

 Are there any way to get the current zoom level?
I was thinking if its not possible them I'll have to monitor the Tile Map URL to extract the zoom level but that would be too much for me. 

Thankyou in advance.

danthony06, 2024-11-19:
Right now there's not a way to resize that I know of, but I think that makes a lot of sense. I think I'd rather have Mapviz scale the markers itself, based on a user setting, than publish the zoom level and have the markers themselves change. If we had the markers re-publish, it would change it for all listeners of the message, which doesn't seem great.

joeldevasia, 2024-11-21:
So are there any work arounds, atleast for my particular case?

joeldevasia, 2024-11-22:
Okay thanks.

danthony06, 2026-09-20:
THe scaling functionality was recently changed, so this doesn't apply any more.

closed 2026-09-20 by danthony06