swri-robotics/mapviz #846: I can not get tile map from Stadia

raminzare17, 2025-06-03:
it did not show map, I used "https://tiles.stadiamaps.com/tiles/stamen_terrain/{z}/{x}/{y}.png?api_key=YOUR-API-KEY" with my API code that i got from stadia account.
I tested link in browers and it shows me png file,
Can problem from {z} {x} {y} that mapviz calculates?

DangitBen, 2025-06-16:
Mapviz doesn't use the `{z}` format. Instead it uses `{zoom}`. Could you try again with the following url and your API key:
`https://tiles.stadiamaps.com/tiles/stamen_terrain/{zoom}/{x}/{y}.png?api_key=YOUR-API-KEY`

stevedanomodolor, 2025-06-27:
I am trying to use the staia terrna maps but with not success even after getting and api key. Please am I doinhg something wrong. 

[image]

My config

capture_directory: "~"
fixed_frame: map
target_frame: map
fix_orientation: false
rotate_90: false
enable_antialiasing: true
show_displays: true
show_status_bar: true
show_capture_tools: true
window_width: 2494
window_height: 1011
view_scale: 1.77156043
offset_x: -55.0469894
offset_y: -4.65194941
use_latest_transforms: true
background: "#a0a0a4"
image_transport: raw
displays:
  - type: mapviz_plugins/tile_map
    name: new display
    config:
      visible: true
      collapsed: false
      custom_sources:
        - base_url: https://tiles.stadiamaps.com/tiles/stamen_terrain/{level}/{x}/{y}.png?api_key=XXXXXXXXXXXXXX
          max_zoom: 15
          name: Stadia
          type: wmts
      bing_api_key: ""
      source: Stadia
  - type: mapviz_plugins/point_click_publisher
    name: new display
    config:
      visible: true
      collapsed: false
      topic: clicked_point
      output_frame: wgs84
  - type: mapviz_plugins/tf_frame
    name: new display
    config:
      visible: true
      collapsed: false
      frame: base_link
      color: "#00ff00"
      draw_style: arrows
      position_tolerance: 0
      buffer_size: 1
      static_arrow_sizes: true
      arrow_size: 53
  - type: mapviz_plugins/navsat
    name: new display
    config:
      visible: true
      collapsed: false
      topic: /gps/fix
      color: "#55aaff"
      draw_style: points
      position_tolerance: 0
      buffer_size: 1
      qos_depth: 10
      qos_history: 1
      qos_reliability: 1
      qos_durability: 2

stevedanomodolor, 2025-06-27:
1. I am running with a launch file 
2. Yes, It says Ok after i put in the link with api key
3. 3. 
1. Latest version on the repo, I installed from source since there is an issue with rolling binaries. see https://github.com/swri-robotics/mapviz/issues/852
2. Ubuntu 24

This is my launch file

```
import launch
import launch.actions
import launch.substitutions
import launch_ros.actions
import os
from ament_index_python.packages import get_package_share_directory

gps_wpf_dir = get_package_share_directory("nav2_gps_waypoint_follower_demo")
mapviz_config_file = os.path.join(gps_wpf_dir, "config", "gps_wpf_demo.mvc")

def generate_launch_description():
    return launch.LaunchDescription([
        launch_ros.actions.Node(
            package="mapviz",
            executable="mapviz",
            name="mapviz",
            parameters=[{"config": mapviz_config_file}]
        ),
        launch_ros.actions.Node(
            package="swri_transform_util",
            executable="initialize_origin.py",
            name="initialize_origin",
            remappings=[
                ("fix", "gps/fix"),
            ],
        ),
        launch_ros.actions.Node(
            package="tf2_ros",
            executable="static_transform_publisher",
            name="swri_transform",
            arguments=["0", "0", "0", "0", "0", "0", "map", "origin"]
        )
    ])

```

For more code context, see
https://github.com/stevedanomodolor/navigation2_tutorials/tree/feat/update_gps_tutorial/nav2_gps_waypoint_follower_demo/launch

DangitBen, 2025-06-30:
Just an update @stevedanomodolor:

- Your config worked for me with `mapviz` on the rolling version and my OS on Ubuntu  `22.04`
- I am working on getting an environment to test rolling on `24.04 `(I haven't had much time to do this)
- I don't suspect there is much different between `24.04` and `22.04` but I will report back.
- If everything work for me, then you will need to provide more detailed information about your machine and any logs that mapviz outputs.

DangitBen, 2025-06-30:
@stevedanomodolor Following those instructions everything worked for me, with one adjustment.

I changed the `api_key=xxxxx` in the config to my `api_key` _before_ running `ros2 launch mapviz mapviz.launch.py`. Sometimes the `tile_map` plugin will not try to reload tiles if your authentication fails or some other network error occurs. If you do edit your `api_key` with mapviz open, simply save the config and reopen mapviz.

stevedanomodolor, 2025-06-30:
this is my full log
root@stevedan:/map_ws# ros2 launch mapviz mapviz.launch.py
[INFO] [launch]: All log files can be found below /root/.ros/log/2025-06-30-19-48-05-922612-stevedan-8549
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [mapviz-1]: process started with pid [8552]
[INFO] [initialize_origin.py-2]: process started with pid [8553]
[INFO] [static_transform_publisher-3]: process started with pid [8554]
[static_transform_publisher-3] [WARN] [1751312885.997844922] []: Old-style arguments are deprecated; see --help for new-style arguments
[static_transform_publisher-3] [INFO] [1751312886.020823373] [swri_transform]: Spinning until stopped - publishing transform
[static_transform_publisher-3] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-3] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-3] from 'map' to 'origin'
[mapviz-1] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[initialize_origin.py-2] [INFO] [1751312886.309506401] [initialize_origin]: Origin: swri
[initialize_origin.py-2] [INFO] [1751312886.310016059] [initialize_origin]: Frame: map
[initialize_origin.py-2] [INFO] [1751312886.324050240] [initialize_origin]: Origin from 'manual' source set to 29.45196669, -98.61370577, 233.719
[mapviz-1] [INFO] [1751312886.772793725] [mapviz]: Target frame selected: <none>
[mapviz-1] [INFO] [1751312886.839502327] [mapviz]: View scale: 1.000000 meters/pixel
[mapviz-1] [INFO] [1751312886.903687547] [mapviz]: Subscribing to /local_xy_origin
[mapviz-1] [INFO] [1751312886.906524936] [mapviz]: Found mapviz plugin: mapviz_plugins/attitude_indicator
[mapviz-1] [INFO] [1751312886.906557920] [mapviz]: Found mapviz plugin: mapviz_plugins/coordinate_picker
[mapviz-1] [INFO] [1751312886.906566141] [mapviz]: Found mapviz plugin: mapviz_plugins/disparity
[mapviz-1] [INFO] [1751312886.906572841] [mapviz]: Found mapviz plugin: mapviz_plugins/draw_polygon
[mapviz-1] [INFO] [1751312886.906579308] [mapviz]: Found mapviz plugin: mapviz_plugins/float
[mapviz-1] [INFO] [1751312886.906585883] [mapviz]: Found mapviz plugin: mapviz_plugins/gps
[mapviz-1] [INFO] [1751312886.906591870] [mapviz]: Found mapviz plugin: mapviz_plugins/grid
[mapviz-1] [INFO] [1751312886.906598424] [mapviz]: Found mapviz plugin: mapviz_plugins/image
[mapviz-1] [INFO] [1751312886.906604439] [mapviz]: Found mapviz plugin: mapviz_plugins/laserscan
[mapviz-1] [INFO] [1751312886.906610582] [mapviz]: Found mapviz plugin: mapviz_plugins/marker
[mapviz-1] [INFO] [1751312886.906616892] [mapviz]: Found mapviz plugin: mapviz_plugins/measuring
[mapviz-1] [INFO] [1751312886.906622929] [mapviz]: Found mapviz plugin: mapviz_plugins/multires_image
[mapviz-1] [INFO] [1751312886.906628898] [mapviz]: Found mapviz plugin: mapviz_plugins/navsat
[mapviz-1] [INFO] [1751312886.906634849] [mapviz]: Found mapviz plugin: mapviz_plugins/occupancy_grid
[mapviz-1] [INFO] [1751312886.906640773] [mapviz]: Found mapviz plugin: mapviz_plugins/odometry
[mapviz-1] [INFO] [1751312886.906647644] [mapviz]: Found mapviz plugin: mapviz_plugins/path
[mapviz-1] [INFO] [1751312886.906654042] [mapviz]: Found mapviz plugin: mapviz_plugins/plan_route
[mapviz-1] [INFO] [1751312886.906659974] [mapviz]: Found mapviz plugin: mapviz_plugins/point_click_publisher
[mapviz-1] [INFO] [1751312886.906665684] [mapviz]: Found mapviz plugin: mapviz_plugins/pointcloud2
[mapviz-1] [INFO] [1751312886.906671790] [mapviz]: Found mapviz plugin: mapviz_plugins/pose
[mapviz-1] [INFO] [1751312886.906677601] [mapviz]: Found mapviz plugin: mapviz_plugins/robot_image
[mapviz-1] [INFO] [1751312886.906683237] [mapviz]: Found mapviz plugin: mapviz_plugins/route
[mapviz-1] [INFO] [1751312886.906688910] [mapviz]: Found mapviz plugin: mapviz_plugins/string
[mapviz-1] [INFO] [1751312886.906695426] [mapviz]: Found mapviz plugin: mapviz_plugins/textured_marker
[mapviz-1] [INFO] [1751312886.906702027] [mapviz]: Found mapviz plugin: mapviz_plugins/tf_frame
[mapviz-1] [INFO] [1751312886.906708061] [mapviz]: Found mapviz plugin: mapviz_plugins/tile_map
[mapviz-1] [INFO] [1751312886.912893167] [mapviz]: Loading configuration from /root/.mapviz_config
[mapviz-1] [INFO] [1751312886.913656319] [mapviz]: fixed frame selected: map
[mapviz-1] [INFO] [1751312886.913729657] [mapviz]: Target frame selected: map
[mapviz-1] [INFO] [1751312886.916949895] [mapviz]: creating: mapviz_plugins/tile_map
[mapviz-1] [WARN] [1751312886.963620282] [mapviz]: [transform_manager]: No transformer from 'wgs84' to 'map'. If 'map' is a /tf frame, it may not have been broadcast recently.
[mapviz-1] [ERROR] [1751312886.963701010] [mapviz]: Error: No transform between wgs84 and map
[mapviz-1] [INFO] [1751312886.963788122] [mapviz]: Reorder displays
[mapviz-1] [INFO] [1751312886.969150047] []: LocalXyWgs84Util initializing origin to lat: 29.451967, lon: -98.613706, alt: 233.719000
[mapviz-1] [INFO] [1751312887.037972654] [mapviz]: OK
[mapviz-1] [ERROR] [1751312887.219224706] [mapviz]: NETWORK ERROR: Host requires authentication
[mapviz-1] [ERROR] [1751312887.245504303] [mapviz]: NETWORK ERROR: Host requires authentication
[mapviz-1] [ERROR] [1751312887.252352546] [mapviz]: NETWORK ERROR: Host requires authentication
[mapviz-1] [ERROR] [1751312887.265012429] [mapviz]: NETWORK ERROR: Host requires authentication
[mapviz-1] [ERROR] [1751312887.270318204] [mapviz]: NETWORK ERROR: Host requires authentication
[... cut ...]
[mapviz-1] [ERROR] [1751312888.030506760] [mapviz]: NETWORK ERROR: Host requires authentication
[mapviz-1] [ERROR] [1751312888.057951886] [mapviz]: NETWORK ERROR: Host requires authentication
[mapviz-1] [ERROR] [1751312888.088709428] [mapviz]: NETWORK ERROR: Host requires authentication
[mapviz-1] [ERROR] [1751312888.114455562] [mapviz]: NETWORK ERROR: Host requires authentication
[mapviz-1] [ERROR] [1751312888.141928415] [mapviz]: NETWORK ERROR: Host requires authentication
[mapviz-1] [ERROR] [1751312888.169664637] [mapviz]: NETWORK ERROR: Host requires authentication

stevedanomodolor, 2025-06-30:
> [@stevedanomodolor](https://github.com/stevedanomodolor) Following those instructions everything worked for me, with one adjustment.
>
> I changed the `api_key=xxxxx` in the config to my `api_key` [...]

i tried what you proposed but still dont see the map, maybe the authentication process takes time?

DangitBen, 2025-06-30:
Okay, then this is something you need to work through on that end! Good luck!

closed 2025-06-30 by DangitBen