swri-robotics/mapviz #851: Bing maps are no more available

mattiadutto, 2025-06-13:
What are your plans for substituting Bing Maps, since they will no longer be available in a couple of weeks? It is also no longer possible to create a new Bing Maps account. I'm asking because if I select Staman map I have the following error: `NETWORK ERROR: Host tile.stamen.com not found`.

DangitBen, 2025-06-13:
I think the stamen maps have just moved to a new url. I think we could still support those by just updating that.

I'll have to look around to see if there are any other map sources we'd want to support by default, now that Bing is not an option.

DangitBen, 2025-06-13:
1. Yes, you are correct. It seems Stamen no longer hosts tiles themselves, and they suggest [migrating to Stadia Maps](https://maps.stamen.com/stadia-partnership/) as you found. I have not tried this and won't have time today to get to it.
2. I am aware of that tutorial. I **cannot recommend it** as that goes against Google Maps's [Terms of Service](https://cloud.google.com/maps-platform/terms).
3. No one here has tried Azure Maps yet, but I will take your word for it. Hosting tile servers can be expensive and most companies will want to charge if you use enough to escape their free tier.

### In the mean time:

You have a couple options.

#### Externally hosted maps

OpenStreetMaps _seems_ to have a hosted version of their data as tiles, but we'd have to update to be in compliance with their [usage requirements](https://operations.osmfoundation.org/policies/tiles/). And if it gets used too heavily, there are no guarantees that it will stay reliable/available. You could try using this for the tile maps source in mapviz and let me know if it works:
`https://tile.openstreetmap.org/{level}/{x}/{y}.png`

**For US users** who need imagery, I recommend using something like USGS for a public domain imagery source. You can see the sources available [here](https://basemap.nationalmap.gov/arcgis/rest/services) or simply try this URL:
`https://basemap.nationalmap.gov/arcgis/rest/services/USGSImageryOnly/MapServer/WMTS/tile/1.0.0/USGSImageryOnly/default/default028mm/{level}/{y}/{x}.png`

For those outside the US, your country or other agencies may provide freely available sources with their own licensing. These may be outdated though.
E.g. The European Union seems to have [WMS/WMTS services](https://data.europa.eu/data/datasets/cartes-topographiques-services-wms-et-wmts?locale=en) and you could try this URL as an example:
`https://wmts1.geoportail.lu/opendata/wmts/basemap/GLOBAL_WEBMERCATOR/{level}/{x}/{y}.png`

QGIS or [GDAL](https://gdal.org/en/stable/drivers/raster/wmts.html) are your friend here. Use them to explore and try out different sources. I could provide a guide, but that is pretty far outside the scope of this repository.

#### Locally hosted maps

See [this guide](https://swri-robotics.github.io/mapviz/guides/local_tile_map_imagery/) about using tiles data stored locally on the drive.