swri-robotics/mapviz #857: Problem ROS2 Humble: no map and no GPS received

rdouguet, 2025-08-25:
Hello,

I would like to display a trace on a map using mapviz, but I have encountered two problems.

I am running **ROS2 Humble on Ubuntu 22.04**, and I installed all mapviz tools correctly. I can launch mapviz with:

`ros2 run mapviz mapviz`

**1st problem: No map displayed**

I set the **Fixed Frame** and **Target Frame** to wgs84.
I tested different Stamen sources, but the result is always a gray map.

I also tried a **Custom WMTS Source** with:
- Base URL: https://tile.openstreetmap.org/{level}/{x}/{y}.png
- Max Zoom: 19

But I still only get a gray background.

**2nd problem: No messages received**

I am publishing a navsat topic named /xtf_streamer/gps_fix.
When I check with:

`ros2 topic echo /xtf_streamer/gps_fix --once`

I can see the following data:
```
header:
  stamp:
    sec: 1750145174
    nanosec: 180000000
  frame_id: wgs84
status:
  status: 0
  service: 1
latitude: 47.884578983164174
longitude: -4.659755203425984
altitude: -25.692548751831055
position_covariance:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
position_cov
```
The QoS is set to History: Keep Last, Reliability: Reliable, Durability: Volatile, but in mapviz the status of this topic stays “No message received.”

I also attach a screenshot of the result in mapviz.
Do you have any ideas about what might be wrong?

I will test on another PC tomorrow to see if I get the same result.

Thanks in advance!

[image]

DangitBen, 2025-08-26:
## Gray Map

> I tested different Stamen sources, but the result is always a gray map.

Those Stamen maps are no longer available by the map provider.

> I set the Fixed Frame and Target Frame to wgs84.

`tile_map` is designed to work with a frame in unit meters (`wgs84` is in longitude, latitude; not meters). Try setting an origin with a frame id, then switching your fixed and target frames to that. You can test  if this solves the gray map issue for you by launching mapviz with the provided launch file rather than running it alone:
 `ros2 launch mapviz mapviz.launch.py`

This sets up the `origin` and `map` frames, which can be used as your `Fixed Frame` with `tile_map`. You can copy that launch file into your workspace/package and edit it so that it can use your mapviz config, or just load your config after the application opens with the `File > Open Config` dialog.

## No messages received

I'm not certain from the information that you have provided why the subscriber is not receiving those messages. I'd check to make sure that your publisher _and_ subscriber (mapviz) are [compatible](https://docs.ros.org/en/rolling/Concepts/Intermediate/About-Quality-of-Service-Settings.html#qos-compatibilities) and will result in the expected behavior. If you are confident your topics are compatible and mapviz is providing no terminal/rosout errors, then would you be able to provide a small bag with messages that reproduce this error?

DangitBen, 2025-08-26:
I am able to display see the data from the bag that you provided if I launch with an origin set up (i.e. `ros2 launch mapviz mapviz.launch.py`). The NavSatFix plugin/messages inherently require an origin to be set, since all of their data is in a geocentric frame like `wgs84` (by the way, your `frame_id` in your bag was set to `map` and that is probably not quite correct but not a big deal here). I think when mapviz doesn't have an origin set, it skips displaying the message silently. This is a case where mapviz should display a better status message on the plugin, rather than claiming it has no messages.

[image]

>  I was thinking it might be useful to also print an info message in the NavSatFixCallback (e.g. "Message received OK") at least the first time, to confirm reception.

Once it can transform and display the message, the plugin will display "OK". This is shown in the screenshot.

> My messages are timestamped with the original recording time. Could that be causing the points not to be displayed? Or maybe the points are displayed too briefly for me to notice?

mapviz considers the transform from `wgs84` to your origin to be a static transform (unchanging over time), so it will happily display old data without an issue. Additionally, the NavSatFix plugin doesn't have a duration/lifetime setting, so it keeps the messages as long as the plugin stays active.

Example launch file I used:
```python
import launch
import launch.actions
import launch.substitutions
import launch_ros.actions

def generate_launch_description():
    return launch.LaunchDescription(
        [
            launch_ros.actions.Node(
                package="mapviz",
                executable="mapviz",
                name="mapviz",
            ),
            launch_ros.actions.Node(
                package="swri_transform_util",
                executable="initialize_origin.py",
                name="initialize_origin",
                parameters=[
                    {"local_xy_frame": "map"},
                    {"local_xy_origin": "rdouguet"},
                    {
                        "local_xy_origins": """[
                    {"name": "rdouguet",
                        "latitude": 48.284581363857505,
                        "longitude": -4.659414738435398,
                        "altitude": -25.683584213256836,
                        "heading": 0.0},
                ]""" # I just grabbed one of the points from your bag
                    },
                ],
            ),
            launch_ros.actions.Node(
                package="tf2_ros",
                executable="static_transform_publisher",
                name="swri_transform",
                arguments=["0", "0", "0", "0", "0", "0", "map", "origin"],
            ),
        ]
    )
```

rdouguet, 2025-08-27:
Thanks for your feedback.

Actually, my mistake wasn't due to incorrect initialization of the origin, but rather to the /tf commands, which I probably don't fully understand. When in doubt, I copied your launch exactly and replayed the bag I sent you yesterday.

Attached is the terminal feedback (I do have a no transform wgs84 and map error).

> ros2 launch launcher start_mapviz.launch.py 
> ...
> [INFO] [launch]: Default logging verbosity is set to INFO
> [INFO] [mapviz-1]: process started with pid [30020]
> [INFO] [initialize_origin.py-2]: process started with pid [30022]
> [INFO] [static_transform_publisher-3]: process started with pid [30024]
> [static_transform_publisher-3] [WARN] [1756287406.239461472] []: Old-style arguments are deprecated; see --help for new-style arguments
> [static_transform_publisher-3] [INFO] [1756287406.246803157] [swri_transform]: Spinning until stopped - publishing transform
> [static_transform_publisher-3] translation: ('0.000000', '0.000000', '0.000000')
> [static_transform_publisher-3] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
> [static_transform_publisher-3] from 'map' to 'origin'
> [mapviz-1] [INFO] [1756287406.445114607] [mapviz]: Target frame selected: <none>
> [initialize_origin.py-2] [INFO] [1756287406.502019617] [initialize_origin]: Origin: rdouguet
> [initialize_origin.py-2] [INFO] [1756287406.502219164] [initialize_origin]: Frame: map
> [initialize_origin.py-2] [INFO] [1756287406.506416689] [initialize_origin]: Origin from 'manual' source set to 48.284581363857505, -4.659414738435398, -25.683584213256836
> ...
> [mapviz-1] [INFO] [1756287406.552467156] [mapviz]: Loading configuration from /home/douguet/.mapviz_config
> [mapviz-1] [INFO] [1756287406.552832699] [mapviz]: fixed frame selected: map
> [mapviz-1] [INFO] [1756287406.552859343] [mapviz]: Target frame selected: map
> [mapviz-1] [INFO] [1756287406.554112674] [mapviz]: creating: mapviz_plugins/tile_map
**> [mapviz-1] [WARN] [1756287406.599459422] [mapviz]: [transform_manager]: No transformer from 'wgs84' to 'map'. If 'map' is a /tf frame, it may not have been broadcast recently.
> [mapviz-1] [ERROR] [1756287406.599611282] [mapviz]: Error: No transform between wgs84 and map**
> [mapviz-1] [INFO] [1756287406.599654335] [mapviz]: Reorder displays
> [mapviz-1] [INFO] [1756287406.600650850] [mapviz]: creating: mapviz_plugins/navsat
> [mapviz-1] [INFO] [1756287406.656229313] [mapviz]: Toggle show plugin
> [mapviz-1] [INFO] [1756287406.656281460] [mapviz]: Reorder displays
> [mapviz-1] [WARN] [1756287406.657149458] [mapviz]: No messages received.
> [mapviz-1] [INFO] [1756287406.663215771] [mapviz]: Subscribing to /xtf_streamer/gps_fix
> [mapviz-1] [INFO] [1756287406.665172943] []: LocalXyWgs84Util initializing origin to lat: 48,284581, lon: -4,659415, alt: -25,683584
> [mapviz-1] [INFO] [1756287406.698111497] [mapviz]: OK

I'm trying to follow the documentation to implement this plugin, but I admit that I don't see what else to change to get the correct /tf values. I'm attaching a screenshot of the configuration and the /tf values, because I think that's where my error is coming from...

[image]

[image]

closed 2025-08-27 by rdouguet